Moddable-OpenSource/moddable #22: Subclasses of SMBus or I2C cannot have read() or write() methods

rwaldron, 2017-12-21:
Given the following: 

```js
import SMBus from "pins/smbus";

class Component extends SMBus {
  constructor(dictionary) {
    super(dictionary);
  }
  read(register) {
    return this.readByte(register);
  }
  write(register, byte) {
    this.writeByte(register, byte);
  }
}
```

Any attempt to call read() or write() will result in the following: 

```
stack overflow (-1)!
unhandled exception
```

phoddie, 2017-12-21:
I think you can solve that in `smbus.js` by using `super.read` and `super.write` in place of `this.read` and `this.write`.

rwaldron, 2018-01-16:
I'd like to work on this this week, but I just hit #28 and cannot build/deploy code anymore :\

phoddie, 2018-01-18:
Latest revision of [`smbus.js`](https://github.com/Moddable-OpenSource/moddable/blob/80aea6d5e550d994ad8cfc0792b7862f89930a96/modules/pins/smbus/smbus.js#L29) calls `read` and `write` through `super` so subclasses can safely have their own `read` and `write` functions.

closed 2018-01-18 by phoddie