Moddable-OpenSource/moddable #24: How to use the SSD1306 driver?

rwaldron, 2017-12-21:
I tried to figure this out, but I'm lost! Is there any existing example code + manifest.json for using SSD1306? Thanks!!

phoddie, 2017-12-21:
I put together a trivial example, but don't have a way to test it with me. Here's the manifest:
`````
{
	"include": [
		"$(MODDABLE)/examples/manifest_base.json",
		"$(MODDABLE)/examples/manifest_commodetto.json",
		"$(MODDABLE)/modules/pins/i2c/manifest.json"
	],
	"modules": {
		"*": [
			"./main",
			"$(MODULES)/drivers/ssd1306/*",
			"$(MODULES)/pins/digital/esp/*",
			"$(MODULES)/pins/spi/esp/*",
		]
	},
	"defines": {
		"ssd1306": {
			width: 128,
			height: 32,
		}
	},
	"config": {
		"screen": "ssd1306",
	},
	"preload": [
		"ssd1306"
	]
}
`````

If the display is not 128x32, update the `defines` section of the manifest.

Here's an trivial app:

`````
import SSD1306 from "ssd1306";
import Poco from "commodetto/Poco";

let poco = new Poco(screen);

poco.begin();
	poco.fillRectangle(poco.makeColor(255, 255, 255), 0, 0, poco.width, poco.height);
poco.end();
`````
The `ssd1306` can be used over SPI or I2C. The default is SPI, so some adjustments are needed to work with I2C. Happy to put that together if needed.

Once the display is working, more or less any of the Commodetto graphics examples apply - though you don't have very many pixels to work with.

The SSD1306 driver requires the input pixels to be 8 bit gray, as this is easier to map to black and white and more than enough color resolution for a 1-bit display. When building with `mcconfig`, set the pixel format to gray256:

`````
mcconfig -d -m -p esp -f gray256
`````

rwaldron, 2018-01-04:
Thanks, I should have some time to try this out next week

phoddie, 2018-01-06:
There are now [docs](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/drivers/ssd1306/ssd1306.md) for the SSD1306 as well as an [example](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/drivers/ssd1306) that shows how to use the [dither](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/drivers/ssd1306/ssd1306.md#dither) option.

closed 2018-01-06 by phoddie