Moddable-OpenSource/moddable #25: Commit 2e323e5 breaks esp target on Windows

wdimmit, 2017-12-31:
The following change to `manifest_base.json` breaks the esp target under Windows.  When broken, `mcconfig -x -m -p esp` returns `### Error: (host): No such file or directory`

				`"c++11": [
					"*.cc.o",
					"*.cpp.o",
				],`

Removing the addition allows mcconfig to work as expected on commit db3216e (current HEAD).  I'm not sure what the other ramifications of this change might be.

This was tested with several examples, including piu/list.

bfriedkin, 2018-01-01:
Will -

Thank you for reporting the issue. The problem is that the windows build is missing the c++11 build recipe for the esp platform. To workaround the issue, please clone the %MODDABLE%\tools\mcconfig\make.esp.c++11.mk file to nmake.esp.c++11.mk in the same directory. We will push a fix soon.

- Brian

phoddie, 2018-01-06:
Resolved by [commit](https://github.com/Moddable-OpenSource/moddable/commit/4683b6849e8189e4ee9b97421d936fcf7646722f#diff-6b7a521eed811ba9590a7d6599505a86) from @bfriedkin.

closed 2018-01-06 by phoddie