Moddable-OpenSource/moddable #26: I2C and separation of concerns

fivdi, 2018-01-05:
On an ESP32 each time an I2C driver object is instantiated the I2C controller hardware on the ESP32 is reconfigured. For example, the following code creates two I2C driver objects and each object wishes to set the bus speed to a different value (whether or not this makes sense is a different question but it's possible):
```js
import LIS3DH from "lis3dh";
import MCP23017 from "MCP230XX";

const lis3dh = new LIS3DH({hz: 400000});
const mcp23017 = new MCP23017({hz: 100000});
```
During instantiation [modI2CInit](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/pins/i2c/esp32/modI2C.c#L32-L44) is called for each object and the I2C controller is reconfigured each time by calling [i2c_param_config](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/pins/i2c/esp32/modI2C.c#L42). The bus speed is set to 400000Hz the first time `i2c_param_config` is called and to 100000Hz the second time `i2c_param_config` is called.

There appears to be a separation of concerns issue here. On the one hand there is an I2C bus and on the other hand there are I2C devices that are connected to an I2C bus. Both of these abstractions are being rolled into a single abstraction here.

The above analysis is based on reading the code and I apologize if it is incorrect. I haven't tried anything out because the ESP32 is only supported macOS and I only have a Linux machine.

phoddie, 2018-01-06:
@fivdi, thank you for the careful review. You are correct that the current ESP32 I2C implementation won't behave as correctly if there are devices operating at different speeds. The `modI2CActivate` function should reinitialize the bus when necessary when switching to a device operating at a bus speed different from the current bus speed. The [ESP8266 implementation of `modI2CActivate`](https://github.com/Moddable-OpenSource/moddable/blob/5f140532461807673071fc99d66b6e79327a1083/modules/pins/i2c/esp/modI2C.c#L61) does this correctly. I'll look at updating the ESP32 implementation to match.

fivdi, 2018-01-06:
@phoddie thank you for the fast response. I hadn't looked at the ESP8266 implementation.

To ensure that I understand everything correctly I have a few followup questions.

According to LIS3DH datasheet its maximum SCL clock frequency is 400 kHz and according to the MCP23017 datasheet its maximum SCL clock frequency is 1.7 MHz. 

Lets say the I2C bus should be clocked at the maximum possible frequency. The following code is not correct because it will result in the LIS3DH being clocked at 1.7 MHz when the system is communicating with the MCP23017. Is this Correct?

```js
import LIS3DH from "lis3dh";
import MCP23017 from "MCP230XX";

const lis3dh = new LIS3DH({hz: 400000});
const mcp23017 = new MCP23017({hz: 1700000});
```

To run the bus at the maximum possible frequency the lowest maximum frequency of all devices connected to the bus should be determined. In this case that would be 400 kHz. This frequency must then be passed to all driver objects during instantiation. Something like the following code is needed. Is this correct?

```js
import LIS3DH from "lis3dh";
import MCP23017 from "MCP230XX";

const lis3dh = new LIS3DH({hz: 400000});
const mcp23017 = new MCP23017({hz: 400000});
```

If this is taken a step further and the pins to use for SCL and SDA were not the defaults the code could look something like this:

```js
import LIS3DH from "lis3dh";
import MCP23017 from "MCP230XX";

const lis3dh = new LIS3DH({hz: 400000, scl: 11, sda: 12});
const mcp23017 = new MCP23017({hz: 400000, scl: 11, sda: 12});
```

At the end of the day an object that describes how the bus is configured is passed to the constructor for each device on that bus. In general, the bus will be configured in the same way for each device on the bus so why is it necessary to tell each device about the configuration?

After writing this I now realize that I should learn about manifests to see if that's a more appropriate place for specifying how the bus should be configured.

phoddie, 2018-01-08:
There is no explicit API for the I2C bus. It is implicit based on the configuration of the active I2C instances.

The ESP8266 implementation of I2C changes the I2C communication speed to match the settings of the device on the current I2C transaction. The ESP32 implementation will be updated to this behavior (changes already made).

If each device uses the same I2C settings, use a single settings object to invoke the constructors. This eliminates repetition. The example above becomes:

````javascript
const settings = {hz: 400000, scl: 11, sda: 12};
const lis3dh = new LIS3DH(settings);
const mcp23017 = new MCP23017(settings);
````

phoddie, 2018-01-18:
FYI - we've added support to build for ESP32 on Linux. The [build instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20–%20Getting%20Started.md#esp32-setup-1) for Linux have been updated with the details.

fivdi, 2018-01-18:
> FYI - we've added support to build for ESP32 on Linux.

Excellent, thank you very much.

> The build instructions for Linux have been updated with the details.

The build instructions are very good. I just tried them out and everything worked flawlessly the first time. Also, it was possible to get blinky up and running in the debugger on a HUZZAH32 without reading any Moddable SDK documentation. The only documentation I had to take a look at was the HUZZAH32 documentation to find out which GPIO the LED is connected to. Nice work :+1:

fivdi, 2018-01-18:
It looks like the issue related to bus speed mentioned in the OP has been resolved.

> There is no explicit API for the I2C bus. It is implicit based on the configuration of the active I2C instances.

This isn't how I saw things when looking at the documentation and code initially. I saw a class called [SMBus](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/pins/smbus/smbus.js#L29) that extends a class called I2C which is [documented as providing access to the I2C bus](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/pins/pins.md#class-i2c). I assumed that an instance of class called SMBus should be SMBus. In reality this is not the case. An instance of SMBus represents a device connected to an SMBus. At least that's the way I see things now. This doesn't bother me to the extent that it did when I opened this issue so I'll close the issue now.

closed 2018-01-18 by fivdi