Moddable-OpenSource/moddable #30: Build error on Mac

autonome, 2018-01-24:
Got my Moddable Zero 😍

But I'm getting a build error:

MacBook-2:mac dietrich$ make
make -f /Users/dietrich/misc/moddable/xs/makefiles/mac/xsc.mk
make -f /Users/dietrich/misc/moddable/xs/makefiles/mac/xsid.mk
make -f /Users/dietrich/misc/moddable/xs/makefiles/mac/xsl.mk
make -f serial2xsbug.mk
make -f simulator.mk
2018-01-22 22:38:41.916 ibtoold[30827:2942346] [MT] DVTPlugInLoading: Failed to load code for plug-in com.apple.dt.IDE.IDEInterfaceBuilderCocoaTouchIntegration (/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework), error = Error Domain=NSCocoaErrorDomain Code=3587 "dlopen_preflight(/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration): Library not loaded: /Library/Developer/PrivateFrameworks/CoreSimulator.framework/Versions/A/CoreSimulator
  Referenced from: /Applications/Xcode.app/Contents/Developer/Library/PrivateFrameworks/SimulatorKit.framework/Versions/A/SimulatorKit
  Reason: image not found" UserInfo={NSLocalizedFailureReason=The bundle is damaged or missing necessary resources., NSLocalizedRecoverySuggestion=Try reinstalling the bundle., NSFilePath=/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration, NSDebugDescription=dlopen_preflight(/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration): Library not loaded: /Library/Developer/PrivateFrameworks/CoreSimulator.framework/Versions/A/CoreSimulator
  Referenced from: /Applications/Xcode.app/Contents/Developer/Library/PrivateFrameworks/SimulatorKit.framework/Versions/A/SimulatorKit
  Reason: image not found, NSBundlePath=/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework, NSLocalizedDescription=The bundle “IDEInterfaceBuilderCocoaTouchIntegration” couldn’t be loaded because it is damaged or missing necessary resources.}, dyldError = dlopen(/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration, 0): Library not loaded: /Library/Developer/PrivateFrameworks/CoreSimulator.framework/Versions/A/CoreSimulator
  Referenced from: /Applications/Xcode.app/Contents/Developer/Library/PrivateFrameworks/SimulatorKit.framework/Versions/A/SimulatorKit
  Reason: image not found
2018-01-22 22:38:41.918 ibtoold[30827:2942346] [MT] DVTPlugInExtensionFaulting: Failed to fire fault for extension Xcode.InterfaceBuilderKit.AppleTVIntegration.Singletons: Error Domain=DVTPlugInErrorDomain Code=2 "Loading a plug-in failed." UserInfo={DVTPlugInIdentifierErrorKey=com.apple.dt.IDE.IDEInterfaceBuilderCocoaTouchIntegration, DVTPlugInExecutablePathErrorKey=/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration, NSLocalizedRecoverySuggestion=The plug-in or one of its prerequisite plug-ins may be missing or damaged and may need to be reinstalled., DVTPlugInDYLDErrorMessageErrorKey=dlopen(/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration, 0): Library not loaded: /Library/Developer/PrivateFrameworks/CoreSimulator.framework/Versions/A/CoreSimulator
  Referenced from: /Applications/Xcode.app/Contents/Developer/Library/PrivateFrameworks/SimulatorKit.framework/Versions/A/SimulatorKit
  Reason: image not found, NSLocalizedDescription=Loading a plug-in failed., NSFilePath=/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework, NSLocalizedFailureReason=The plug-in “com.apple.dt.IDE.IDEInterfaceBuilderCocoaTouchIntegration” at path “/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework” could not be loaded.  The plug-in or one of its prerequisite plug-ins may be missing or damaged., NSUnderlyingError=0x7fd8797278f0 {Error Domain=NSCocoaErrorDomain Code=3587 "dlopen_preflight(/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration): Library not loaded: /Library/Developer/PrivateFrameworks/CoreSimulator.framework/Versions/A/CoreSimulator
  Referenced from: /Applications/Xcode.app/Contents/Developer/Library/PrivateFrameworks/SimulatorKit.framework/Versions/A/SimulatorKit
  Reason: image not found" UserInfo={NSLocalizedFailureReason=The bundle is damaged or missing necessary resources., NSLocalizedRecoverySuggestion=Try reinstalling the bundle., NSFilePath=/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration, NSDebugDescription=dlopen_preflight(/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework/IDEInterfaceBuilderCocoaTouchIntegration): Library not loaded: /Library/Developer/PrivateFrameworks/CoreSimulator.framework/Versions/A/CoreSimulator
  Referenced from: /Applications/Xcode.app/Contents/Developer/Library/PrivateFrameworks/SimulatorKit.framework/Versions/A/SimulatorKit
  Reason: image not found, NSBundlePath=/Applications/Xcode.app/Contents/PlugIns/IDEInterfaceBuilderCocoaTouchIntegration.framework, NSLocalizedDescription=The bundle “IDEInterfaceBuilderCocoaTouchIntegration” couldn’t be loaded because it is damaged or missing necessary resources.}}}
2018-01-22 22:38:41.923 ibtoold[30827:2942346] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'Error getting value for parameter key 'name' of extension 'Xcode.InterfaceBuilderKit.AppleTVIntegration.Singletons' in plug-in 'com.apple.dt.IDE.IDEInterfaceBuilderAppleTVIntegration''
*** First throw call stack:
(
	0   CoreFoundation                      0x00007fff3d7d800b __exceptionPreprocess + 171
	1   libobjc.A.dylib                     0x00007fff64461c76 objc_exception_throw + 48
	2   DVTFoundation                       0x000000010825052d -[DVTExtensionParameter valueForKey:error:] + 0
	3   IDEInterfaceBuilderKit              0x00000001079dcbed -[IBScopedSingletonRegistry init] + 722
	4   IDEInterfaceBuilderKit              0x00000001079dc901 __43+[IBScopedSingletonRegistry sharedInstance]_block_invoke + 41
	5   libdispatch.dylib                   0x00007fff65017d50 _dispatch_client_callout + 8
	6   libdispatch.dylib                   0x00007fff65017d03 dispatch_once_f + 41
	7   IDEInterfaceBuilderKit              0x00000001079dc8d5 +[IBScopedSingletonRegistry sharedInstance] + 45
	8   IDEInterfaceBuilderKit              0x00000001077f3d05 -[IBInterfaceBuilderPlugin init] + 146
	9   IDEInterfaceBuilderKit              0x00000001077f3c59 +[IBInterfaceBuilderPlugin ide_initializeWithOptions:error:] + 34
	10  IDEFoundation                       0x0000000108b7cf42 _IDEInitializeOnePlugInAndPrerequisites + 2324
	11  IDEFoundation                       0x0000000108b7c849 _IDEInitializeOnePlugInAndPrerequisites + 539
	12  IDEFoundation                       0x0000000108b7c849 _IDEInitializeOnePlugInAndPrerequisites + 539
	13  IDEFoundation                       0x0000000108b7af3e _IDEInitializePlugIns + 1604
	14  IDEFoundation                       0x0000000108b7a2f5 IDEInitialize + 8547
	15  ibtoold                             0x00000001074ef170 ibtoold + 74096
	16  ibtoold                             0x00000001074ee2eb ibtoold + 70379
	17  libdyld.dylib                       0x00007fff65051115 start + 1
)
libc++abi.dylib: terminating with uncaught exception of type NSException
make[1]: *** [/Users/dietrich/misc/moddable/build/bin/mac/debug/Screen Test.app/Contents/Resources/English.lproj/main.nib] Abort trap: 6
make: *** [debug] Error 2

bfriedkin, 2018-01-24:
Dietrich -

Sorry to hear that you are having trouble with the build. The build requires macOS Sierra version 10.12 or newer and Xcode version 9 or newer. Could you please confirm the versions you are using? For reference, I am using macOS Sierra version 10.12.6 and Xcode version 9.2.

Assuming your versions are up-to-date, before building the Moddable tools, did you launch Xcode and install the additional required components when prompted? The build errors suggest that there are missing components.

- Brian

autonome, 2018-01-24:
Bah, I did the xcode CLI bit, but never launched the app itself! Got the
additional components prompt - installing and will try again. Thanks Brian!

autonome, 2018-01-24:
Yep, that did the trick, thanks!

closed 2018-01-24 by autonome