Moddable-OpenSource/moddable #31: Reading the ambient temperature register from an MCP9808 results in an error

fivdi, 2018-01-25:
The goal of the following program is to read the ambient temperature register from an MCP9808 temperature sensor.
```js
import Timer from "timer";
import Digital from "pins/digital";
import SMBus from "pins/smbus";

let count = 0;
let mcp9808 = new SMBus({address: 0x18});

Timer.repeat(() => {
  Digital.write(13, ~count & 1);

  let rawTemp = mcp9808.readWord(0x05);

  trace(`repeat ${++count} ${rawTemp}\n`);
}, 200);
```
When run, the program results in the following error:
```
/home/brian/src/moddable/modules/pins/i2c/i2c.c (156) # Break: (host): write file!
```
Line 156 in i2c.c is [here](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/pins/i2c/i2c.c#L156).

I think the problem here is that the [SMBus readWord method](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/pins/smbus/smbus.js#L34-L38) implements an SMBus transaction as two I2C transactions, i.e., a write transaction followed by a read transaction. The MCP9808 only appears to work if the write is followed by a "repeat start" and then a read. See sample code on [page 25 of the MCP9808 datasheet](http://ww1.microchip.com/downloads/en/DeviceDoc/25095A.pdf).

I have verified that the MCP9808 is connected up correctly and can successfully read the ambient temperature register with program implemented in C using the Espressif IoT Development Framework.

phoddie, 2018-01-26:
@fivdi, thanks for the report. It is possible that the problem is in `SMBus.prototype.readWord` implementation and that would be easy enough to fix. However, that particular error usually means there's no I2C communication occurring. May I ask which pins you are using for `SCL` and `SDA` on the ESP32?

fivdi, 2018-01-26:
> May I ask which pins you are using for SCL and SDA on the ESP32?

You may indeed ask this question but unfortunately the answer reveals the depths of my stupidity. SCL is pin 22 and SDA is pin 23. I did however neglect to mention this anywhere and somehow expected it to magically function correctly.

The following code works fine:
```js
import Timer from "timer";
import Digital from "pins/digital";
import SMBus from "pins/smbus";

let count = 0;
let mcp9808 = new SMBus({address: 0x18, sda: 23, scl: 22});

Timer.repeat(() => {
  Digital.write(13, ~count & 1);

  let word = mcp9808.readWord(0x05);
  let rawTemp = (word >> 8) + ((word & 0xff) << 8);
  let celsius = (rawTemp & 0x0fff) / 16;

  if (rawTemp & 0x1000) {
    celsius -= 256;
  }

  trace(`repeat ${++count} ${celsius}°C\n`);
}, 1000);
```

I'm not sure whether or not implementing `SMBus.prototype.readWord` as two I2C transactions is always going to work but that's a different question and isn't why this issue was created.

Thanks for the help and sorry for the noise.

closed 2018-01-26 by fivdi

phoddie, 2018-01-26:
Glad it was that was an easy fix. FWIW, I do intend to extend the I2C API to include write-then-read as a single transaction.