Moddable-OpenSource/moddable #35: linux build needs echo -e

pmcneil, 2018-01-30:
in tools.mk the echo lines include escape codes like \n. Mainly echo '#!bin/bash\nsome command'
gnu echo needs a -e flag to interpret the escapes, so the build fails with command not found when it tries to run mcconfig et al.

bfriedkin, 2018-01-31:
Peter -

Thank you for letting us know about this issue. It seems that "gnu echo" behavior with escape codes can vary by platform.

As a potentially more portable alternative, some developers have suggested using "printf" instead of "echo -e". That seemed to work for us in tools.mk on both macOS and Ubuntu Linux 16.04. Could you please try the same on your development host and let us know if that resolves the issue?

Thanks,
Brian

pmcneil, 2018-02-01:
Brian,
will try (I seem to have deleted by install accidentailly, so I have to try it again anyway :-) )

pmcneil, 2018-02-03:
I can confirm that printf is a good solution

bfriedkin, 2018-10-19:
The Linux [tools.mk](https://github.com/Moddable-OpenSource/moddable/blob/public/build/makefiles/lin/tools.mk) file was updated to replace echo with printf.

Regards,
Brian

closed 2018-10-19 by bfriedkin