Moddable-OpenSource/moddable #41: Code error

eco747, 2018-03-04:
In the file xsAll.h:

there is a strange code in mxIsFunction, mxIsArray, mxIsString, mxIsBoolean, mxIsNumber, mxIsDate, mxIsRegExp, mxIsSymbol, mxIsHost, mxIsProxy

```
(/* (THE_SLOT) && */ ((THE_SLOT)->next) && ((THE_SLOT)->next->flag && XS_INTERNAL_FLAG) && ((THE_SLOT)->next->kind == XS_STRING_KIND) || ((THE_SLOT)->next->kind == XS_STRING_X_KIND))
look at the && instead of &

(THE_SLOT)->next->flag && XS_INTERNAL_FLAG)
                       ^^ here

```

patrick-soquet, 2018-03-05:
Terrible typo indeed. Luckily enough, the test is currently redundant since it is always true for matching kinds (XS_CALLBACK_KIND, etc). Fix committed. The open source tree will be updated soon. Thank you much.

phoddie, 2018-04-21:
Fix integrated

closed 2018-04-21 by phoddie