Moddable-OpenSource/moddable #42: Memory crash on exceptions in timer callback

eco747, 2018-03-11:
module/base/timer/modTimer.c : xs_timer_callback function should take care of exceptions in callback call because in case of exception fxForget is not called but c_free is.

more generally exceptions are not handled in timer callback and that not really good. They should also be traced somewhere.

code should be 
```
        xsBeginHost(ts->the);

		xsTry {
			xsCallFunction1(xsReference(ts->callback), xsGlobal, ts->self);
		}
		xsCatch {
			//xsStringValue message = fxToString(ts->the,&xsException);
			//xsDebugLog( message );
		}

		if (0 == modTimerGetSecondInterval(timer)) {
			xsForget(ts->self);
			ts->callback = NULL;
		}

	xsEndHost(ts->the);
```

phoddie, 2018-03-11:
Good catch. It may be enough to reorganize the code as follows:

	xsBeginHost(ts->the);
		xsCallFunction1(xsReference(ts->callback), xsGlobal, ts->self);
	xsEndHost(ts->the);

	if (0 == modTimerGetSecondInterval(timer)) {
		xsForget(ts->self);
		xsmcSetHostData(ts->self, NULL);
	}

An unhanded exception stops at `xsEndHost`. The implementation of [xsEndHost logs](https://github.com/Moddable-OpenSource/moddable/blob/cf07a0f4c378d32e504eeaa0007a8cd2a1468707/xs/includes/xs.h#L1164) unhanded exceptions.

eco747, 2018-03-12:
good choice

phoddie, 2018-04-21:
Fix integrated

closed 2018-04-21 by phoddie