Moddable-OpenSource/moddable #43: Distinguishing reference types and iterating over object properties

iamtakingiteasy, 2018-03-24:
Hi, i am trying to write native printing function, which takes any kind of JS value and prints it out, recursing into arrays and objects, also wrapping strings into quotes.

This is what i came up with:

```C

#include <xs/xsAll.h>
#include <xs.h>
#include <stdio.h>

void hinarin_root_print(xsMachine *the) {
    for (xsIntegerValue i = 0; i < xsToInteger(xsArgc); i++) {
        if (i > 0) {
            printf(" ");
        }
        txSlot *instance = &xsArg(i);
        txSlot *reference;
        switch (xsTypeOf(xsArg(i))) {
            case xsReferenceType:
                reference = instance->value.reference;
                switch (reference->flag) { // where to find flag values for this field? also, is this a correct test on reference type?
                    case 0: // looks like object
                        printf("{");
                        for (txSlot *next = reference->next; next; next = next->next) { // is this a correct way to iterate over object properties?
                            if (next != reference->next) {
                                printf(", ");
                            }
                            printf("%s: ", xsName(next->ID));
                            xsCallFunction1(xsGet(xsGlobal, xsID("print")), xsNull, *next);
                        }
                        printf("}");
                        break;
                    case 1: // looks like array
                        printf("[");
                        for (txIndex d = 0; d < reference->next->value.array.length; d++) {
                            if (d > 0) {
                                printf(", ");
                            }
                            xsCallFunction1(xsGet(xsGlobal, xsID("print")), xsNull, reference->next->value.array.address[d]);
                        }
                        printf("]");
                        break;
                    case 2: // looks like function
                        printf("function() { ... }");
                        break;
                    default: // don't know what it is
                        printf("unknown");
                        break;
                }
                break;
            case xsStringType:
                printf("\"%s\"", instance->value.string);
                break;
            default:
                printf("%s", xsToString(*instance));
                break;
        }
    }
}

void hinarin_root_println(xsMachine *the) {
    hinarin_root_print(the);
    printf("\n");
}

void hinarin_bind_root(xsMachine *the) {
    xsDefine(xsGlobal, xsID("print"), xsNewHostFunction(hinarin_root_print, 1), xsChangeAll);
    xsDefine(xsGlobal, xsID("println"), xsNewHostFunction(hinarin_root_println, 1), xsChangeAll);
}
```

but I have a few questions, which are outlined in comments to code, thanks.
P.S. Any hints on how to do common things and what can be done better with XS6 are also quite welcome.

patrick-soquet, 2018-03-26:
Here under is a simple implementation. I hope it will help. A few comments:
- Do not include `xsAll.h`, that is the private API, include `xs.h`, that is the public API.
- The documentation of the public API is coming soon. Sorry for the delay. 
- Use `xsToString` to get strings from primitive values.
- Use `xsEnumerate` to get an iterator on enumerable properties, like `for in`.
- There is a static to recurse only 3 times but of course you can have auxiliary functions to change the limit, etc.

```
#include "xs.h"

static int global_print_depth = 0;
void global_print(xsMachine* the)
{
	xsIntegerValue c = xsToInteger(xsArgc), i;
	xsVars(4);
	global_print_depth++;
	for (i = 0; i < c; i++) {
		if (i)
			printf(" ");
		switch (xsTypeOf(xsArg(i))) {
		case xsUndefinedType:
		case xsNullType:
		case xsBooleanType:
		case xsIntegerType:
		case xsNumberType:
			printf("%s", xsToString(xsArg(i)));
			break;
		case xsSymbolType:
			xsResult = xsCall1(xsGlobal, xsID("String"), xsArg(i));
			printf("%s", xsToString(xsResult));
			break;
		case xsStringType:
		case xsStringXType:
			printf("'%s'", xsToString(xsArg(i)));
			break;
		case xsReferenceType:
			if (global_print_depth < 3) {
				xsBooleanValue comma = 0;
				printf("{");
				xsVar(0) = xsEnumerate(xsArg(i));
				for (;;) {
					xsVar(1) = xsCall0(xsVar(0), xsID("next"));
					if (xsTest(xsGet(xsVar(1), xsID("done"))))
						break;
					xsVar(2) = xsGet(xsVar(1), xsID("value"));
					xsVar(3) = xsGetAt(xsArg(i), xsVar(2));
					if (comma)
						printf(",");
					else
						comma = 1;
					printf(" %s:", xsToString(xsVar(2)));
					xsCall1(xsGlobal, xsID("print"), xsVar(3));
				}
				printf(" }");
			}
			else
				printf("%s", xsToString(xsArg(i)));
			break;
		}
	}
	global_print_depth--;
	if (!global_print_depth)
		printf("\n");
}
```

iamtakingiteasy, 2018-03-26:
But without xsAll.h, how to tell `{}` object from `[]` array and them two from functions()? `xsTypeOf` reports that they are all `xsReferenceType`.

iamtakingiteasy, 2018-03-26:
What does `X` in xsStringXType mean? (I think the same goes for CODE/CODE_X kinds, which are part of internal API, which difference is also unclear).

phoddie, 2018-03-26:
Once you know it is a reference, you can use `xsIsInstanceOf`. For example, to determine if it is a function or array:

```
if (xsIsInstanceOf(xsVar(0), xsFunctionPrototype))
	;
if (xsIsInstanceOf(xsArg(1), xsArrayPrototype))
	;
```

closed 2018-06-06 by phoddie