Moddable-OpenSource/moddable #50: Build errors on Linux

fivdi, 2018-05-01:
I cloned moddable about 30 minutes ago and get errors while building on Ubuntu 14.04:
```
# tools debug : cc tools
/home/brian/src/moddable/build/tmp/lin/debug/tools/commodettoPoco.c.o: In function `xs_poco_build':
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:114: undefined reference to `CFENew'
/home/brian/src/moddable/build/tmp/lin/debug/tools/commodettoPoco.c.o: In function `xs_poco_begin':
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:194: undefined reference to `CFELockCache'
/home/brian/src/moddable/build/tmp/lin/debug/tools/commodettoPoco.c.o: In function `xs_poco_end':
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:281: undefined reference to `CFELockCache'
/home/brian/src/moddable/build/tmp/lin/debug/tools/commodettoPoco.c.o: In function `xs_poco_getTextWidth':
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:670: undefined reference to `CFESetFontData'
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:681: undefined reference to `CFEGetGlyphFromUnicode'
/home/brian/src/moddable/build/tmp/lin/debug/tools/commodettoPoco.c.o: In function `xs_poco_drawText':
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:721: undefined reference to `CFESetFontData'
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:724: undefined reference to `CFEGetGlyphFromUnicode'
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:745: undefined reference to `CFEGetGlyphFromUnicode'
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:770: undefined reference to `CFEGetGlyphFromUnicode'
/home/brian/src/moddable/modules/commodetto/commodettoPoco.c:780: undefined reference to `CFEGetGlyphFromUnicode'
collect2: error: ld returned 1 exit status
make[1]: *** [/home/brian/src/moddable/build/bin/lin/debug/tools] Error 1
make[1]: Leaving directory `/home/brian/src/moddable/build/makefiles/lin'
make: *** [debug] Error 2
```
The complete output of the build can be seen [here](https://gist.github.com/fivdi/29856f0afd0b5ab13cb84b77ab71a16b).

Moddable-OpenSource, 2018-05-01:
_fixing_

fivdi, 2018-05-01:
Thank you very much. Build works now.

closed 2018-05-01 by fivdi