Moddable-OpenSource/moddable #51: MacOS ESP8266 (Moddable Zero) setup doesnt verify

sanandnarayan, 2018-05-02:
Hi, I have the MacOS getting started guide successfully till step 8 of ESP8266 (Moddable Zero) setup. I was able to verify the host environment setup by building the starter helloworld application for the desktop simulator target.

But when  I do step 9 of ESP8266 (Moddable Zero) setup
I get the following error 

`make: *** No rule to make target /Users/anand/esp/ESP8266_RTOS_SDK/lib/libcirom.a', needed by /Users/anand/code/playground/moddable/build/tmp/esp/debug/lib/lib_a-setjmp.o'.  Stop`

i dont find the lib folder inside `/Users/anand/esp/ESP8266_RTOS_SDK`

May i know what could be the issue and how I could debug this. Thank you for the support

fivdi, 2018-05-04:
@sanandnarayan I'm not sure if you've noticed but commit a5dc8ae7eca1d7e7f2dc54e8f84b1a3585a9f8cb which references this issue looks like it should fix the issue.

@mkellner I'm not sure if you expected the reference from commit a5dc8ae7eca1d7e7f2dc54e8f84b1a3585a9f8cb to this issue to trigger notifications for this issue but if so this is not the case.

sanandnarayan, 2018-05-04:
Thank you for the support. It works 👍 :-)

closed 2018-05-04 by sanandnarayan