Moddable-OpenSource/moddable #52: The httpserver example stops working or crashes

fivdi, 2018-05-02:
On an ESP32 the [httpserver](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/http/httpserver/main.js) example stops working after responding to a relatively small number of requests from a single client. The same example crashes after responding to small number of requests from two clients that send requests at the same time.

#### Case 1: The httpserver example stops working after responding to a relatively small number of requests from a single client

If the httpserver example is running on an ESP32 and the following bash script if used as a client from a terminal window:
```bash
#!/bin/sh
c=0
while [ 1 ]
do
    curl http://192.168.1.14/"$c"
    echo
    sleep .1
    c=$((c+1))
done
```
then the httpserver stops working after responding to a little over 100 requests. Here's the output from the client in the terminal window:
```
hello, client at path /0.
hello, client at path /1.
hello, client at path /2.
... unnecessary output removed here
hello, client at path /124.
hello, client at path /125.
hello, client at path /126.
curl: (7) Failed to connect to 192.168.1.14 port 80: Connection refused

curl: (7) Failed to connect to 192.168.1.14 port 80: Connection refused

curl: (7) Failed to connect to 192.168.1.14 port 80: Connection refused

... unnecessary output removed here

curl: (7) Failed to connect to 192.168.1.14 port 80: Connection refused

curl: (7) Failed to connect to 192.168.1.14 port 80: Connection refused

curl: (7) Failed to connect to 192.168.1.14 port 80: Connection refused

```
#### Case 2: The httpserver example crashes after responding to small number of requests from two clients that send requests at the same time

If the httpserver example is running on an ESP32 and the following bash script if used as a client from two terminal windows at the same time:
```bash
#!/bin/sh
c=0
while [ 1 ]
do
    curl http://192.168.1.14/"$c"
    echo
    sleep .1
    c=$((c+1))
done
```
then the httpserver crashes. Here's the output from the first client in the first terminal window:
```
hello, client at path /0.
hello, client at path /1.
hello, client at path /2.
hello, client at path /3.
hello, client at path /4.
hello, client at path /5.
hello, client at path /6.
hello, client at path /7.
hello, client at path /8.
hello, client at path /9.
hello, client at path /10.
hello, client at path /11.
hello, client at path /12.
hello, client at path /13.
hello, client at path /14.
hello, client at path /15.
hello, client at path /16.
hello, client at path /17.
hello, client at path /18.
```
Here's the output from the second client in the second terminal window:
```
hello, client at path /0.
hello, client at path /1.
hello, client at path /2.
hello, client at path /3.
hello, client at path /4.
hello, client at path /5.
hello, client at path /6.
hello, client at path /7.
hello, client at path /8.
hello, client at path /9.
hello, client at path /10.
hello, client at path /11.
hello, client at path /12.
hello, client at path /13.
hello, client at path /14.
hello, client at path /15.
```
And here's the crash report:
```
Guru Meditation Error: Core  0 panic'ed (LoadProhibited)
. Exception was unhandled.
Core 0 register dump:
PC      : 0x4013a7b9  PS      : 0x00060d30  A0      : 0x800ee710  A1      : 0x3ffbcaa0  
A2      : 0x3ffd621c  A3      : 0x3ffb7f8c  A4      : 0x00190000  A5      : 0x000019e9  
A6      : 0x00000000  A7      : 0x0000111c  A8      : 0x8013a7b6  A9      : 0x3ffbca80  
A10     : 0x000019e9  A11     : 0x3ffd621c  A12     : 0x00000000  A13     : 0x3ffb7dac  
A14     : 0x3ffd7b2c  A15     : 0x80808080  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000010  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  
Backtrace: 0x4013a7b9:0x3ffbcaa0 0x400ee70d:0x3ffbcac0 0x400dfbe2:0x3ffbcba0 0x400d4b8f:0x3ffbcc00 0x400db18b:0x3ffbcc20 0x400dfbe2:0x3ffbcc40 0x400daf6a:0x3ffbcca0 0x400dfc8d:0x3ffbccc0 0x400d4b8f:0x3ffbcd20 0x400d4d35:0x3ffbcd40 0x400ed0bd:0x3ffbcd60 0x400edafe:0x3ffbce70 0x400d34b7:0x3ffbce90 0x400d2461:0x3ffbcec0
Entering gdb stub now.
```
#### Additional Information
To ensure this is not a hardware issue the [arduino-esp32 SimpleWiFiServer](https://github.com/espressif/arduino-esp32/blob/master/libraries/WiFi/examples/SimpleWiFiServer/SimpleWiFiServer.ino) was tested with two clients that send request at the same time. The bash script shown above was used for both clients. The SimpleWiFiServer responded successfully to tens of thousands of requests from both clients without any issues.

phoddie, 2018-05-04:
@fivdi - Thanks for the detailed report. I'm certain (2) is a consequence of cross-task calls to lwip, which isn't safe with the lwip raw interface. It doesn't appear when running the same code on ESP8266/Adruino because there's no preemption there. I'll clean this up. The cause of (1) is less clear, it may be the same. We'll know after fixing (2).

phoddie, 2018-06-06:
FYI - currently testing changes to the socket implementation which allow a large number (1000+) of back-to-back http client requests to run reliably. Going to apply the same approach to the server related portions of socket.

phoddie, 2018-06-09:
Changes committed to lwip/modSocket.c to deal with multi-threaded execution on ESP32. I've successfully tested both http client and server with 1000+ transactions without a crash or memory  leak. Please give it a try. 

That said, this stuff is tricky enough that there may well still be problems. I observed two rare issues when stress testing:

- I suspect there's still a problem lurking when an accepted incoming socket is closed by the remote. 
- I've seen the ESP32 Wi-Fi stack run out of buffers after a large number of rejected attempts to connect from ESP32 to a remote.

closed 2018-06-19 by phoddie

fivdi, 2018-07-15:
Sorry about the delay in answering. I performed a few tests with the latest version of the Moddable SDK.

Below are the results of four consecutive tests. All tests were performed with one client.

With the first test there was a crash after 164 successful requests:
```
Leaving...
Hard resetting via RTS pin...
Backtrace: 0x4008e94c:0x3ffbc7d0 0x4008eae7:0x3ffbc7f0 0x4008e395:0x3ffbc810 0x40082276:0x3ffbc830 0x400827e1:0x3ffbc850 0x4000bec7:0x3ffbc870 0x400ee572:0x3ffbc890 0x400ed61a:0x3ffbc8b0 0x400ede52:0x3ffbc8d0 0x400d34bf:0x3ffbc8f0 0x400d247d:0x3ffbc920
Entering gdb stub now.
```
With the second test there was a crash after 280 successful requests:
```
Leaving...
Hard resetting via RTS pin...
Backtrace: 0x4008e94c:0x3ffbc7d0 0x4008eae7:0x3ffbc7f0 0x4008e395:0x3ffbc810 0x40082276:0x3ffbc830 0x400827e1:0x3ffbc850 0x4000bec7:0x3ffbc870 0x400ee572:0x3ffbc890 0x400ed61a:0x3ffbc8b0 0x400ede52:0x3ffbc8d0 0x400d34bf:0x3ffbc8f0 0x400d247d:0x3ffbc920
Entering gdb stub now.
```
For the third test the delay between requests was increased to 0.5 seconds. There was a crash after 290 successful requests:
```
Leaving...
Hard resetting via RTS pin...
Backtrace: 0x4008e94c:0x3ffbc7d0 0x4008eae7:0x3ffbc7f0 0x4008e395:0x3ffbc810 0x40082276:0x3ffbc830 0x400827e1:0x3ffbc850 0x4000bec7:0x3ffbc870 0x400ee572:0x3ffbc890 0x400ed61a:0x3ffbc8b0 0x400ede52:0x3ffbc8d0 0x400d34bf:0x3ffbc8f0 0x400d247d:0x3ffbc920
Entering gdb stub now.
```
With the fourth test there was a crash after 128 successful requests:
```
Leaving...
Hard resetting via RTS pin...
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40139daf  PS      : 0x00060b30  A0      : 0x800ee550  A1      : 0x3ffbc870  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x8008b224  A5      : 0x3ffb9a30  
A6      : 0x00000003  A7      : 0x00060023  A8      : 0x00000007  A9      : 0x3ffbc9ac  
A10     : 0x00000001  A11     : 0x00000000  A12     : 0x8008b224  A13     : 0x3ffb9a30  
A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000038  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  
Backtrace: 0x40139daf:0x3ffbc870 0x400ee54d:0x3ffbc890 0x400ed61a:0x3ffbc8b0 0x400ede52:0x3ffbc8d0 0x400d34bf:0x3ffbc8f0 0x400d247d:0x3ffbc920
Entering gdb stub now.
```

phoddie, 2018-08-09:
There was a race condition that caused the socket to be disposed twice when both the client and server closed the socket at about the same time. On some networks, the problem happens immediately; on others you can run for 100's of connections without a problem. The current fix has been tested against your batch file on two very different networks with over 15,000 connections on both ESP32 and ESP8266. With some luck, it will work more reliably for you as well.

fivdi, 2018-08-25:
Things have improved a lot and many more requests can now be handled. However, when multiple clients are running at the same time I'm still seeing crashes.

Here's what I see when running the batch file twice at the the same time.

ESP32 output:
```
Leaving...
Hard resetting via RTS pin...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x400ee3b2  PS      : 0x00060930  A0      : 0x800eeb75  A1      : 0x3ffbc900  
A2      : 0xadce7c00  A3      : 0x3ffccc60  A4      : 0x3ffb0874  A5      : 0xa8c07369  
A6      : 0x00000000  A7      : 0x00000001  A8      : 0x06ff0000  A9      : 0x3ffbc8e0  
A10     : 0x3ffb0874  A11     : 0xadce7c28  A12     : 0x00060920  A13     : 0x3ffbc950  
A14     : 0x3ffccc60  A15     : 0x40080000  SAR     : 0x00000000  EXCCAUSE: 0x0000001d  
EXCVADDR: 0xadce7c28  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  
Backtrace: 0x400ee3b2:0x3ffbc900 0x400eeb72:0x3ffbc920 0x400d327f:0x3ffbc940 0x400d22f5:0x3ffbc970
Entering gdb stub now.
```

xsbug output:
```
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
socketClearPending called - NOTHING PENDING?
```

Client 1 output:
```
hello, client at path /0.
hello, client at path /1.
hello, client at path /2.
...
hello, client at path /6370.
hello, client at path /6371.
hello, client at path /6372.
curl: (7) Failed to connect to 192.168.1.7 port 80: Connection timed out

curl: (7) Failed to connect to 192.168.1.7 port 80: No route to host

curl: (7) Failed to connect to 192.168.1.7 port 80: No route to host

curl: (7) Failed to connect to 192.168.1.7 port 80: No route to host
...
```

Client 2 output:
```
hello, client at path /0.
hello, client at path /1.
hello, client at path /2.
...
hello, client at path /4067.
hello, client at path /4068.
hello, client at path /4069.
```

I have a test running with a single client at the moment which has successfully handled over 9000 requests. I'll leave it running for a few hours and report back.

fivdi, 2018-08-25:
The test with a single client has now successfully handled over 40000 requests. The use cases that I'm currently interested in will only have a single client and everything looks good here now. Thank you very much for looking into this and providing the fix.

fivdi, 2018-08-26:
The test with the single client was left running all night and it ended up crashing after a little over 72000 requests.

ESP32 output:
```
Leaving...
Hard resetting via RTS pin...
Backtrace: 0x4008ea3c:0x3ffbc820 0x4008ebfd:0x3ffbc840 0x4008e485:0x3ffbc860 0x400822ae:0x3ffbc880 0x40082819:0x3ffbc8a0 0x4000bec7:0x3ffbc8c0 0x400ef2f2:0x3ffbc8e0 0x400ee3da:0x3ffbc900 0x400eebb6:0x3ffbc920 0x400d327f:0x3ffbc940 0x400d22f5:0x3ffbc970
Entering gdb stub now.
```

phoddie, 2018-08-26:
Glad it is better for you, if not quite perfect. Curious about the multiple clients, as I tested that exact scenario several times on two different networks. The `socketClearPending called - NOTHING PENDING?` trace is probably a good hint - I left that in though I've never seen it trigger myself. I'll make some time to look at this again.