Moddable-OpenSource/moddable #53: instructions on how to implement i2s mems microphone

Thawab, 2018-05-16:
Hello,

I like the project and would love to see it get traction. I saw the documentation of /pins/audioout.md, and want to know if there is any instructions on how to implement the i2s mems mic. I'm planing to hire a freelancer to work on it and make a pull request for this repo if you guys are ok with it.

Thanks

phoddie, 2018-05-16:
@Thawab, as you noted we don't yet have I2S input implemented. We'd be happy to have it. Our licensing [readme](https://github.com/Moddable-OpenSource/moddable/blob/public/licenses/readme.md#contributor-license-agreement) explains our policy on pull requests.

There is I2S output support for ESP32, and that was pretty straightforward to implement using Espressif's I2S sample codes as a model. My suggestion is to tackle the project in two parts. First get the I2S mem mic working with a small native app on ESP32. From there, build the module around it. That way you are only wrestling with one problem at a time.

Probably the biggest challenge will be delivering the recorded buffers to the script. The [XS in C](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/xs/XS%20in%20C.md#xsbeginhost-and-xsendhost) document describes how to make callbacks from native code to script, and it is done in several modules in the Moddable SDK repository, including the I2S output.

We're happy to help with the JavaScript API design and answering questions along the way.

Thawab, 2018-05-16:
cool, got it.

closed 2018-05-16 by Thawab