Moddable-OpenSource/moddable #54: Porting to CortexM

microwavesafe, 2018-05-17:
I'd like to use the XS engine in a project based on an STM32F401RE. Are there any existing ports to the Cortex M4 architecture?

If not are there any examples of a minimal (running precompiled JS on the target) port?

phoddie, 2018-05-17:
The Silicon Labs Gecko family, which is supported, includes Cortex M4 parts. That said, the basic port is largely CPU architecture independent. We don't have a porting starter sample yet. A reasonable place to start is to take the xsPlatform.h and xsHost.c for a similar platform and adapt from there. That is most of what should need to change for "hello world" to run. If you want to use [mcconfig](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/tools/tools.md#mcconfig) to build (all targets currently do), you'll need to adapt that for the target platform as well.

We're happy to help answer questions along the way. Just let us know.

mkellner, 2018-05-17:
The Moddable tools are hosted on Mac, Win and Linux. The core build instructions for the Moddable SDK can be found for the various target platforms in tools/mcconfig - this is where you'll be defining C include paths, defines and compiler options. make.**yourPlatform**.mk is for Mac and Linux hosts and follows the Makefile format, make.**yourPlatform**.mk is for the Windows host build platform and uses the nmake format.

Some necessary target files to port first include xs/includes/xs**yourPlatform**.h xs/platforms/**yourPlatform**/xsPlatform.[ch] and xsHost.c