Moddable-OpenSource/moddable #55: Moddable Zero: Commodetto example "rotated" fails "not configured for requested rotation"

techninja, 2018-05-27:
Just like the title says, requested rotation code:
```javascript
let render = new Poco(screen, { rotation: 90 });
``` 

Seems pretty straightforward, just try to build and run "rotated" on moddable zero.

BTW, great job on this! Incredible work and kit.

techninja, 2018-05-29:
Aha! Thanks so much! I did that and got it rotated just fine. Though I suspect there's something else going on? Only modification to the `mini-drag` example is poco rotation and build rotation set to 90

[image]

Shouldn't the calculations based off render width/height be correctly rotated? Are there other considerations to consider when rotating the screen? The dragging behavior here is also x/y flipped and incorrect.

phoddie, 2018-05-29:
Thanks for the questions. They will help us improve our examples and documentation for others. Good luck with the project!