Moddable-OpenSource/moddable #58: XS: Incorrect string representation of 'NaN'

tevador, 2018-06-11:
On Windows (built with Visual Studio 2017, VC Tools 140):

    xst.exe -e "print(NaN)"
    -NaN

Prints `-NaN` instead of `NaN`. Same result when NaN is the result of an arithmetic operation:

    xst.exe -e "print(void 0 + 1)"
    -NaN

Interestingly, this gives the correct output:

    xst.exe -e "print(-NaN)"
    NaN

On Linux (built with GCC 6.3.0), NaN is represented correctly:

    ./xst -e "print(NaN)"
    NaN

But this is incorrect:

    ./xst -e "print(-NaN)"
    -NaN

NaN should not have a sign. 

These are the relevant sections of the ECMAScript 6 specs: 
* [ToString Applied to the Number Type](http://www.ecma-international.org/ecma-262/6.0/#sec-tostring-applied-to-the-number-type)
* [Unary - Operator](http://www.ecma-international.org/ecma-262/6.0/#sec-unary-minus-operator-runtime-semantics-evaluation)

baryluk, 2018-06-11:
Some, extra test vectors:

".1e-900"
"-2e900"
"1.2e100000000000000000000000000000000000"
"0.000000000000000000000000000000000000000000000000000000000000000000123e50"
"0.000000000000000000000000000000000000000000000000000000000000000000123e-50"
"0.000000000000000000000000000000000000000000000000000000000000000000123e-350"
"0.000000000000000000000000000000000000000000000000000000000000000000123e390"
"0.000000000000000000000000000000000000000000000000000000000000000000123e-256"
"2.2250738585072011e-308"  // Famous PHP #53632 bug.

tevador, 2018-06-15:
Fixed here: https://github.com/tevador/moddable/commit/525c44960812ad4481984739a33bf8c625da0451

closed 2018-06-18 by tevador

baryluk, 2018-06-18:
I hate to be picky. But where are tests?