Moddable-OpenSource/moddable #61: Hardware Refr Design for ESP32 with 2.8" LCD/Touch Panel

mikew67, 2018-06-12:
Was excited to see your hardware examples at the Bay Area Maker Faire, especially the one at the top left of your samples board (since I had a couple ESP32-WROOM dev boards in hand from the IoT Expo the previous week).  Immediately ordered two 2.8" LCD/Touch-Panels (https://www.ebay.com/itm/201950756171, hope these are the right ones) and received them today.

- How to wire up?  (Logically) the same as the instructions for the ESP8266?  At: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/displays/wiring-guide-generic-2.8-CPT-spi.md

- Which example apps are compatible with this ESP32/LCD/Touch-Panel combination?  Compile flags to use?

TIA :-)

Mike

phoddie, 2018-06-12:
@mikew67 - it was a nice surprise seeing you at Maker Faire and having a chance to catch up.

The displays you ordered look correct - in particular the display driver is the usual one. We usually use the NodeMCU-style ESP32 boards (like [this](https://www.amazon.com/HiLetgo®-ESP-WROOM-32-Development-Microcontroller-Integrated/dp/B0718T232Z/ref=sr_1_6). From your notes, I'm not sure if you have the same. We have worked successfully with others. As long as you get the configuration correct and wire to the same pins we use, you should be good.

The wiring guide for our usual ESP32 SPI LCD set-up is [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/displays/wiring-guide-generic-2.4-spi-esp32.md) in our repository. It uses the same driver (docs [here](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/drivers/ili9341/ili9341.md) as on the ESP8266, so all the same apps should run.

The only difference in building is to set the platform to esp32 when calling mcconfig:

    `mcconfig -d -m -p esp32`

Setting up the SDK from Espressif to build for ESP32 is a little more complicated than setting up the ESP8266 build. Many individuals have successfully done it following our set-up [instructions](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20–%20Getting%20Started.md#esp32-setup).

mikew67, 2018-06-12:
Peter,

Thanks for the very prompt reply.

Yes, I've already set up the SDK and built ESP32 example apps, so don't
anticipate difficulty there.

Will let you know how the LCD setup works,

Mike

closed 2018-06-15 by phoddie