Moddable-OpenSource/moddable #63: Missing step in Getting Started/Mac/ESP32 instructions?

mikew67, 2018-06-17:
Folks,
I'm trying to do a clean install of your code on a MacBook Pro running macOS 10.13.5, Xcode 9.4.1, following your instructions in [Moddable SDK -- Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20–%20Getting%20Started.md#esp32-setup)

At the final step: MacOS/ESP32/10. Verify the setup..., I get the following error:
444s-MacBook-Pro:helloworld mikewirth$ mcconfig -d -m -p esp32
\# cc xsPlatform.c (strings in flash)
...
\# cc xsGlobal.c (strings in flash)
In file included from /Users/mikewirth/esp32/esp-idf/components/lwip/include/lwip/lwip/opt.h:45:0,
                 from /Users/mikewirth/esp32/esp-idf/components/lwip/include/lwip/lwip/netdb.h:32,
                 from /Users/mikewirth/Desktop/Repositories/moddable/xs/platforms/esp/xsPlatform.h:96,
                 from /Users/mikewirth/Desktop/Repositories/moddable/xs/sources/xsCommon.h:41,
                 from /Users/mikewirth/Desktop/Repositories/moddable/xs/sources/xsAll.h:41,
                 from /Users/mikewirth/Desktop/Repositories/moddable/xs/sources/xsAll.c:38:
/Users/mikewirth/esp32/esp-idf/components/lwip/include/lwip/port/lwipopts.h:42:24: fatal error: sys/select.h: No such file or directory
\# cc xsJSON.c (strings in flash)
compilation terminated.

The file is actually found at the path: ~/esp32/esp-idf/components/newlib/platform_include/sys/select.h

- Is some environment variable not set correctly?  FYI, here's my PATH:
`444s-MacBook-Pro:helloworld mikewirth$ echo $PATH
/Users/mikewirth/Desktop/Repositories/moddable/build/bin/mac/release:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/Users/mikewirth/esp32/xtensa-esp32-elf/bin:/Users/mikewirth/esp32/esp-idf
`

- I even set IDF_PATH, to no avail:
`444s-MacBook-Pro:~ mikewirth$ echo $IDF_PATH
/Users/mikewirth/esp32/esp-idf
`
- Could this be related to the issue that a stock install of esp-idf SDK is in the ~/esp folder, not the ~/esp32 folder?

- BTW, I'm accumulating suggestions for improvement of your Getting Started guide.  Will forward them to you after getting the solution to this, final issue.

TIA,

Mike

phoddie, 2018-06-19:
Update pushed with change to support the latest IDF

closed 2018-06-19 by phoddie

mikew67, 2018-06-19:
Peter,

Thanks!

Mike