Moddable-OpenSource/moddable #66: Any easy way to tell which touch panel I have?

mikew67, 2018-06-21:
Got a 320x240 LCD display "with touch panel" from [here](https://www.ebay.com/itm/201950756171?rmvSB=true) and connected it to an ESP-WROOM-32 dev board per your instructions (using signal names, not pin numbers; and with the J1 jumper soldered for 3.3v operation).  Display works fine wirth your examples/piu/ sample apps.  But touch panel doesn't respond, using `touch=xpt2046cd` on the mcconfig line.  And, of course, I get an xsbug halt at pit.js, line 57, `let touch = require(config.touch)`

Any easy way to tell what touch panel I have on this LCD module?  Here's a picture of the back of the panel I got.  (Hope the touch panel can be used with your SDK :-)

TIA,

Mike

[image]

phoddie, 2018-06-21:
The touch controller is the part labeled U2. It is almost surely the xpt2046. You shouldn't need to specify that on the command line, as it is the default in the esp32 build. (Your example says `xpt2046cd` - that would fail at the require statement as you are seeing, as it should be `xp52046`)

mikew67, 2018-06-21:
Peter,
Thanks for the quick response.

- Yes, it looks like I have a xpt2046.  Label on the chip (read with a magnifying glass and bright light :-) is HR2046.

- Building with no "touch=..." param on the mcconfig command line does avoid the xsbug breakpoint at the require(config.touch) line.

- But the touch panel is still unresponsive, e.g., with your drag-color app.  Have tried two panels with the same result (same wiring harness though)

- While I'm at it, the flash writes to the ESP32 seem problematic.  I'm getting repeated MD5 compare errors `A fatal error occurred: MD5 of file does not match data in flash!` File MD5 is always the same, flash MD5 varies.  Took five attempts to get drag-color to load.  Have you been seeing this with ESP32s?

Thanks,

Mike

phoddie, 2018-06-21:
That you got past `require(config.touch)` and the following call to instantiate the touch driver (`touch = new touch`) is a good start. But it doesn't tell us that the connection to the touch screen is working. Only `xs_XPT2046_read` reads from SPI.  I suggest running $MODDABLE/examples/drivers/xpt2046calibrate. It traces the raw touch values to the console. Given your results, maybe nothing will trace.

The wiring for touch is complicated by the fact that the SPI bus pins are shared with the display.  You might try only wiring in touch, not the display, and run xpt2046calibrate to see if that changes something.

mikew67, 2018-06-21:
Built xpt2046calibrate. Get a blue screen with a white fiducial near the top left (with the connector at the bottom) of the screen.  Touch panel is unresponsive; no action on the screen nor output to the log window in xsbug.
If you tell me what pins are used for spi output, I can monitor the signals (with a small Salea data analyzer; 8 channel and can decode SPI protocol.  Seems to work with 3.3v logic, even though it's an early model.)
Other diagnostics I can try?

phoddie, 2018-06-21:
The ESP32 seems to be working well with the SPI bus: the screen works. A wiring problem is the most likely explanation at this point. Looking with the logic analyzer seems a good path.  The pin numbers are on the [wiring diagram ](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/displays/wiring-guide-generic-2.4-spi-esp32.md). Here's the relevant bits of the wiring table:
`````
Touch pins      | ESP32 GPIO
-----------------------------
T_DO            | GPIO 12
T_DIn           | GPIO 13
T_CLK           | GPIO 14
T_IRQ           | GPIO 23
T_CS            | GPIO 18
``````

mikew67, 2018-06-21:
Bingo! That was just the info I needed.  Had a wiring error on T_IRQ.  Touch panel now works (as tested by drag-color).  Note:

- Your visual wiring [diagram](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/displays/wiring-guide-generic-2.4-spi-esp32.md) has a problem.  Specifically, the purple wire for T_IRQ **obscures the GPIO number** on the ESP32 module and I misread it as GPIO21 instead of GPIO23 (and didn't double-check it in the text wirelist above the diagram).

- The pin assignments on my ESP-WROOM-32 dev board are physically **different** than your diagram.  I used the signal names instead of PIN numbers, of course.

Thanks for your persistence in helping me sort this out.

Mike

waldopt, 2018-06-21:
— Mike, 

Good point on the T_IRQ wire partially hiding the PIN number. I will fix.

chris

> 
> Bingo! That was just the info I needed. Had a wiring error on T_IRQ. Touch panel now works (as tested by drag-color). Note:
diagram

phoddie, 2018-06-21:
@mikew67 - glad you got it working. Wiring is always a little challenging.

We'll clean-up the the wiring for our board. If you could contribute a wiring diagram for the ESP-WROOM-32 board you are using, we would gladly integrate that into the docs so others can benefit from your work.

closed 2018-06-21 by phoddie