Moddable-OpenSource/moddable #69: Possible nodeHAP port

diogoviannaaraujo, 2018-07-08:
Hi, I've been wondering how hard would it be a nodeHAP port to moddable. Ive some JS background but am fairly new to hardware and stuff. Been playing with BLE on esp32 and was wondering if I could implement a HomeKit accecory.

Ive read that ESpressif SDK has some HomeKit support baked in, but since I suck at C, freeRTOS and such, I was asking myself if it was feasible to port from NodeHAP.

As first notice it looks like its just a strange http server, something I think its possible with moddable tcp stack. Here a comment on HAPServer.js describes a little bit:

> It turns out that the IP-based version of HomeKit's HAP protocol operates over a sort of pseudo-HTTP. Accessories are meant to host a TCP socket server that initially behaves exactly as an HTTP/1.1 server. So iOS devices will open up a long-lived connection to this server and begin issuing HTTP requests. So far, this conforms with HTTP/1.1 Keepalive. However, after the "pairing" process is complete, the connection is expected to be "upgraded" to support full-packet encryption of both HTTP headers and data. This encryption is NOT SSL. It is a customized ChaCha20+Poly1305 encryption layer.
> Additionally, this "HTTP Server" supports sending "event" responses at any time without warning. The iOS device simply keeps the connection open after it's finished with HTTP request/response traffic, and while the connection is open, the server can elect to issue "EVENT/1.0 200 OK" HTTP-style responses. These are typically sent to inform the iOS device of a characteristic change for the accessory (like "Door was Unlocked").

The part that concerns me is that one:

> This encryption is NOT SSL. It is a customized ChaCha20+Poly1305 encryption layer.

I don't have much experience with encryption, would that be a problem on the moddable on esp32?

Thanks for the attention in advance,
Diogo Vianna

wilberforce, 2019-06-13:
Looks like the code you need is all in the util folder - https://github.com/KhaosT/HAP-NodeJS/blob/master/lib/util/eventedhttp.js

However there is a lot of code and it looks quite complex.

phoddie, 2019-07-12:
@diogoviannaaraujo - It is very possible to implement HAP in the Moddable SDK. We had that working in KinomaJS (the precursor to the Moddable SDK) , though I'm not sure it was ever published.

FWIW - The `StreamCipher` in the Moddable SDK implements ChaCha. I don't believe Poly1305 is there, but the code is readily available and should be straightforward to integrate.

If I can help, please let me know.

bartmichu, 2020-05-26:
Hello, did anyone make some progress on this topic?