Moddable-OpenSource/moddable #70: Problem running BLE examples

diogoviannaaraujo, 2018-07-08:
Hi, im having some problems running any code with BLE.

After following all the guidelines on documentation about sdkconfig im getting this message when trying to mcconfig:

```
# cc modBLEServer.c.o
In file included from /Users/diogoviannaaraujo/Git/moddable/modules/network/ble/esp32/modBLEServer.c:27:0:
/Users/diogoviannaaraujo/Git/moddable/modules/network/ble/esp32/modBLEServer.c: In function 'xs_ble_server_initialize':
/Users/diogoviannaaraujo/esp32/esp-idf/components/bt/include/esp_bt.h:61:50: error: static assertion failed: "please enable bluetooth in menuconfig to use bt.h"
 #define BT_CONTROLLER_INIT_CONFIG_DEFAULT() {0}; _Static_assert(0, "please enable bluetooth in menuconfig to use bt.h");
```

Any ideia why that?

bfriedkin, 2018-07-09:
Hello -

The sdkconfig file shipped with the Moddable SDK disables BLE by default to avoid having to build all the BLE components in the ESP-IDF when the host app doesn't require BLE. Sorry for the inconvenience.

In order to enable BLE, you'll need to edit the $MODDABLE/build/devices/esp32/xsProj/sdkconfig file and then rebuild the Moddable app with mcconfig. Please refer to the BLE documentation here for details:

https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/ble/ble.md#esp32platform

Basically you need to set CONFIG_BT_ENABLED=y and one of CONFIG_GATTC_ENABLE/CONFIG_GATTS_ENABLE, depending on whether or not your BLE app implements the client or server role.

After editing the sdkconfig file, you might also need to ensure that the app is completely rebuilt clean. Delete the $MODDABLE/build/tmp/esp32/debug/<app name> directory before running mcconfig. For example, if you are building the ble/advertiser app:

(edit the sdkconfig file options)
rm -rf $MODDABLE/build/tmp/esp32/debug/advertiser
cd $MODDABLE/examples/network/ble/advertiser
mcconfig -d -m -p esp32

We are hoping to simplify the sdkconfig steps in a future release.

Regards,
Brian

diogoviannaaraujo, 2018-07-09:
I had already done the sdkconfig modification part before I tried the first time,
My git diff to the project just have bt enabled, as gatt server is enabled as default:

```
diogoviannaaraujo@Diogos-MBP:~/Git/moddable/examples/network/ble/heart-rate-server$ git diff
diff --git a/build/devices/esp32/xsProj/sdkconfig b/build/devices/esp32/xsProj/sdkconfig
index d56aa3d..a73f03d 100644
--- a/build/devices/esp32/xsProj/sdkconfig
+++ b/build/devices/esp32/xsProj/sdkconfig
@@ -120,7 +120,7 @@ CONFIG_AWS_IOT_SDK=
 #
 # Bluetooth
 #
-CONFIG_BT_ENABLED=n
+CONFIG_BT_ENABLED=y
 CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
 CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
 CONFIG_BTDM_CONTROLLER_HCI_MODE_UART_H4=
```

But I still keep getting the same error, I've downloaded and recloned again idf, toolchain and moddable but seem to have no effect. Tried to remove build temp as you asked but keep getting in the same problem

Ive tried to verbose the mcconfig and it says its including the bt component from the pdf, end of output of mcconfig with verbose on:

```
# cc modBLESM.c.o
/Users/diogoviannaaraujo/esp32/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -D__ets__ -U__STRICT_ANSI__ -DESP32=1  -DmxUseDefaultSharedChunks=1 -DmxRun=1 -DkCommodettoBitmapFormat=7 -DkPocoRotation=0 -DmxDebug=1 -DMODINSTRUMENTATION=1 -DmxInstrument=1 -I/Users/diogoviannaaraujo/Git/moddable/modules/base/instrumentation -I/Users/diogoviannaaraujo/Git/moddable/modules/base/timer -I/Users/diogoviannaaraujo/Git/moddable/modules/network/ble -I/Users/diogoviannaaraujo/esp32/esp-idf/components -I/Users/diogoviannaaraujo/esp32/esp-idf/components/heap/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/driver/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/soc/esp32/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/soc/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/esp32/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/freertos -I/Users/diogoviannaaraujo/esp32/esp-idf/components/freertos/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/freertos/include/freertos -I/Users/diogoviannaaraujo/esp32/esp-idf/components/lwip/include/lwip -I/Users/diogoviannaaraujo/esp32/esp-idf/components/lwip/include/lwip/port -I/Users/diogoviannaaraujo/esp32/esp-idf/components/mbedtls/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/spi_flash/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/vfs/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/tcpip_adapter/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/tcpip_adapter -I/Users/diogoviannaaraujo/esp32/esp-idf/components/bt/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/bt/bluedroid/api/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/bt/bluedroid/api/include/api -I/Users/diogoviannaaraujo/esp32/esp-idf/components/newlib/include -I/Users/diogoviannaaraujo/esp32/esp-idf/components/newlib/platform_include -I/Users/diogoviannaaraujo/Git/moddable/xs/includes -I/Users/diogoviannaaraujo/Git/moddable/xs/sources -I/Users/diogoviannaaraujo/Git/moddable/xs/sources/pcre -I/Users/diogoviannaaraujo/Git/moddable/xs/platforms/esp -I/Users/diogoviannaaraujo/Git/moddable/build/devices/esp32 -I/Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/lib -I/Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/heart-rate-server -c -Os -g -Wno-unused-variable -Wpointer-arith -Wl,-EL -fno-inline-functions -nostdlib -mlongcalls -mtext-section-literals -falign-functions=4 -MMD -fdata-sections -ffunction-sections -fno-jump-tables -fstrict-volatile-bitfields -DWITH_POSIX -DHAVE_CONFIG_H -D BOOTLOADER_BUILD=1 -DESP_PLATFORM -MP -Wno-implicit-function-declaration -std=gnu99 /Users/diogoviannaaraujo/Git/moddable/modules/network/ble/esp32/modBLESM.c -o /Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/heart-rate-server/modBLESM.c.o
In file included from /Users/diogoviannaaraujo/Git/moddable/modules/network/ble/esp32/modBLEServer.c:27:0:
/Users/diogoviannaaraujo/Git/moddable/modules/network/ble/esp32/modBLEServer.c: In function 'xs_ble_server_initialize':
/Users/diogoviannaaraujo/esp32/esp-idf/components/bt/include/esp_bt.h:103:50: error: static assertion failed: "please enable bluetooth in menuconfig to use bt.h"
 #define BT_CONTROLLER_INIT_CONFIG_DEFAULT() {0}; _Static_assert(0, "please enable bluetooth in menuconfig to use bt.h");
                                                  ^
/Users/diogoviannaaraujo/Git/moddable/modules/network/ble/esp32/modBLEServer.c:91:41: note: in expansion of macro 'BT_CONTROLLER_INIT_CONFIG_DEFAULT'
     esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
                                         ^
make: *** [/Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/heart-rate-server/modBLEServer.c.o] Error 1
make: *** Waiting for unfinished jobs....
```

bfriedkin, 2018-07-10:
Hello -

Please add "#define CONFIG_BT_ENABLED 1" to the top of $MODDABLE/build/devices/esp32/sdkconfig.h and try building again. Let me know how that goes.

We are working on simplifying this part of the ESP32 build. Thank you for your patience.

Regards,
Brian

diogoviannaaraujo, 2018-07-10:
Hey, that worked!
And thank you for ur amazing work!

Got curious on where things did not work between the sdkconfig and the sdkconfig.h
Hope in the future I could be the one helping simplify this config part of the build

Moddable-OpenSource, 2018-07-10:
> Basically you need to set CONFIG_BT_ENABLED=y **and one of CONFIG_GATTC_ENABLE/CONFIG_GATTS_ENABLE**, depending on whether or not your BLE app implements the client or server role.

bfriedkin, 2018-07-11:
Glad to hear you are up and running Moddable BLE apps on ESP32. We appreciate your patience and offer to help!

closed 2018-08-02 by phoddie