Moddable-OpenSource/moddable #72: Doesn't build with x64 native tools.

basuke, 2018-07-11:
It should be documented to choose "x86 Native Tools Command Prompt" in step 5 of Windows host environment setup.

barbiani, 2018-12-17:
@basuke Were you able to build on windows 10?

I was able to build the idf examples on msys32, all moddable tools, moddable examples for win, but for esp32 on the x86 Native Tools Command Prompt I get:

C:\msys32\home\future\moddable\examples\helloworld>mcconfig -m -d -p esp32
FC: cannot open C:\MSYS32\HOME\FUTURE\MODDABLE\BUILD\DEVICES\ESP32\XSPROJ\SDKCONFIG.DEFAULTS.PRIOR - No such file or folder
        1 file(s) copied.
The system cannot find the path specified.
NMAKE : fatal error U1077: 'if' : return code '0x1'
Stop.

C:\msys32\home\future\moddable\examples\helloworld>dir C:\MSYS32\HOME\FUTURE\MODDABLE\BUILD\DEVICES\ESP32\XSPROJ\SDKCONFIG.DEFAULTS.PRIOR

 Directory of C:\MSYS32\HOME\FUTURE\MODDABLE\BUILD\DEVICES\ESP32\XSPROJ

12/17/2018  06:28 AM            14,691 sdkconfig.defaults.prior
               1 File(s)         14,691 bytes

C:\msys32\home\future\moddable\examples\helloworld>echo %IDF_PATH%
C:\msys32\esp32\esp-idf

C:\msys32\home\future\moddable\examples\helloworld>echo %BASE_DIR%
C:\msys32\home\future

C:\msys32\home\future\moddable\examples\helloworld>echo %MODDABLE%
C:\msys32\home\future\moddable

bfriedkin, 2019-07-03:
@basuke -

We have updated the [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md) document to use the "x86 Native Tools Command Prompt for VS 2017" in the Windows host environment setup.

Regards,
Brian

closed 2019-07-03 by bfriedkin