Moddable-OpenSource/moddable #74: Add mcconfig parameter to specify USB port and speed

stonehippo, 2018-07-17:
Feature request:

The current makefile driven build process with `mcconfig` seems to fall a little flat when working with different boards. For example, I am working with an [Adafruit ESP8266 Huzzah breakout](https://www.adafruit.com/product/2471). To use`mmconfig` with that board, I edited `make.esp.mk` to change `UPLOAD_SPEED` to 115200 and `UPLOAD_PORT` to `/dev/cu.usbserial-XXXXXXX` (I'm using an FTDI USB-to-serial cable with the board).

I was able to upload the `helloworld` example to the Huzzah with this config (the default upload speed of 921600 caused errors). But it seems like it would be handy to have the ability to override the values in the `mcconfig` make on the command line. Something like:

```
$ mcconfig -d -m -p esp --port /dev/cu.usbserial-XXXXXXX --speed 115200
```

diogoviannaaraujo, 2018-07-20:
Hi im too having some problem using Adafruit's HUZZAH32 with mcconfig.
The port name is ok since its a Slabs USB-UART but the upload fails randomly.

Im guess its because of a different USB-UART chip, adafruits looks a little smaller.

phoddie, 2018-07-20:
@stonehippo - The challenge with providing command line options to `mcconfig` to configure the embedded device connection is that different MCUs handle that very differently. We could make it work for ESP8266, but it wouldn't likely be universally supported across all MCUs. For ESP8266, you can avoid editing the `make.esp.mk` by setting environment variables instead:

``````
export UPLOAD_PORT= /dev/cu.usbserial-XXXXXXX
export UPLOAD_SPEED=115200
``````

Perhaps that's sufficient?

phoddie, 2018-07-20:
@diogoviannaaraujo  - `mcconfig` uses the `esptool.py` from the Espressif IDF to upload to the ESP32. There's nothing special about `mcconfig` performs the upload. With some boards, we have seen `esptool.py` fail randomly on uploads, most commonly the initial connection fails. In one case, it was a USB cable that couldn't reliable handle high-speed transfers. Sometimes disconnecting and reconnecting the USB cable helps.

This issue is best addressed on the [ESP32 IDF repository](https://github.com/espressif/esp-idf). If you do find a solution, please share as we do encounter this problem from time-to-time ourselves. Thanks.

diogoviannaaraujo, 2018-07-20:
Hi @phoddie, thanks for the attention, I've tested with different cables and still no success. But when I use a default Chinese esp32 board it uploads correctly.

Ive also tested the adafruit board with the esptool.py from pip to instal micro python and got everything working just fine, witch made me find everything even more weird.

As soon I have some time ill try again and post back some more information on the errors.

stonehippo, 2018-07-23:
@phoddie Yes, I think using the environment variables is fine. Thanks for the tip!

closed 2018-07-23 by stonehippo

diogoviannaaraujo, 2018-08-25:
Ok, just came back on this.
Adafruit's esp32 board, that have a smaller CP21XX than the Chinese ones also works better on lower speeds.

with default speed I was getting outputs like this after upload:
`Leaving...
Hard resetting via RTS pin...
??
<xsbug><lognn name="main" valueX"XS"/>//xsbug>
^`

export UPLOAD_SPEED=115200 has resolved the instability.

diogoviannaaraujo, 2018-09-03:
Hi,

Im trying to do some uploads on a esp32 dev board and im getting some errors, so im trying to change the speed as mentioned before but the UPLOAD_SPEED variable is being overwritten, when uploading im getting:

```
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 13824 bytes to 9107...
Wrote 13824 bytes (9107 compressed) at 0x00001000 in 0.1 seconds (effective 779.4 kbit/s)...

A fatal error occurred: Timed out waiting for packet header
```

Its weird since I had success changing the speed before (see upper comment), has anything changed in last commits?

Ive tried to change `make.esp.mk` but no success either, should I do some kind of forced re-make?

Thanks

phoddie, 2018-09-03:
As noted above, each platform handles uploading differently. On ESP32, the upload speed is controlled by [`sdkconfig.defaults`](https://github.com/Moddable-OpenSource/moddable/blob/edb1075d7ef37d274ad70ac08d79cb6816d72d17/build/devices/esp32/xsProj/sdkconfig.defaults#L37). Configure that according to the instructions in the ESP32 IDF documentation. For example:

`````
CONFIG_ESPTOOLPY_BAUD_115200B=
CONFIG_ESPTOOLPY_BAUD_230400B=
CONFIG_ESPTOOLPY_BAUD_921600B=
CONFIG_ESPTOOLPY_BAUD_2MB=
CONFIG_ESPTOOLPY_BAUD_OTHER=y
CONFIG_ESPTOOLPY_BAUD_OTHER_VAL=1500000
CONFIG_ESPTOOLPY_BAUD=1500000
`````

The `make.esp.mk` file is for the ESP8266 build. `make.esp32.mk` is the ESP32 build.