Moddable-OpenSource/moddable #75: What is the C API equivalent to `Object.create(prototype)`?

creationix, 2018-08-01:
I'm trying to create 3 object prototypes in C, I have:

```c
xsSlot objectProto = xsNewObject();
xsSlot circleProto = xsNewObject();
xsSlot rectangleProto = xsNewObject();
```

But I want `circleProto` and `rectangleProro` to inherit from `objectProto`.

creationix, 2018-08-01:
Also I see some docs are missing for some interesting APIs.  Are there examples or tests showing how to use `xsNewHostConstructorObject`, `xsNewHostFunctionObject`, and `xsmcNewHostInstance`?

phoddie, 2018-08-01:
We don't usually define objects entirely in C. Instead we define a JavaScript class with methods implemented in C. This is easier to read, write, and maintain. There are many examples of this in the SDK - the [Commodetto Bitmap](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/commodetto/commodettoBitmap.js) object is a typical:

```
class Bitmap @ "xs_Bitmap_destructor" {
	constructor(width, height, format, buffer, offset) @ "xs_Bitmap"
	get width() @ "xs_bitmap_get_width"
	get height() @ "xs_bitmap_get_height"
	get pixelFormat() @ "xs_bitmap_get_pixelFormat"
	get offset() @ "xs_bitmap_get_offset"
	static depth(pixelFormat) @ "xs_bitmap_get_depth";
}
```

The Bitmap is an example of a Host object, a special object kind in XS that has a native constructor and a native destructor. A Host object must be the root of a class hierarchy. A Host object cannot subclass a normal JavaScript object; a normal JavaScript object can extend a Host object. That means you can write this:

```
class FixedSizeBitmap extends Bitmap {
	get width() {return 16;}
	get height() {return 16;}
}
```

...but cannot write this...

```
class Something {}
class Bitmap @ "xs_Bitmap_destructor" extends Something {}
```

Normal JavaScript objects can implement some of their functions in C. For example:

```
class Foo {
	construct() {}
	add(a, b) {return a + b);
	doSomethingInC @ "xs_doSomethingInC";
}
```

Or:

```
class BitmapWithRotate extends Bitmap {
	rotate(degrees) @ "xs_bitmap_rotate"
}
```

Whether you are writing a Host object or a normal JavaScript object with native methods, it is almost always better to define the class in JavaScript to allow XS to build the structure, then you need only implement the native functions. That's why we haven't document the calls you note above (and, in fact, almost never use in our code... maybe we should remove them from `xs.h`).

creationix, 2018-08-01:
So I'm trying to expose a C library to the JavaScript.  The C library simulates inheritance.  My simple example here with `Shape`, `Rectangle` and `Circle` is a minimal test to figure it out.

Now regarding the workflow, I'm not using the entire moddable SDK, I'm just trying to embed XS into an existing program.  I think I can probably use `xsc` to generate some header files than I then link into my program, I'm still figuring out how that works.

phoddie, 2018-08-01:
That detail helps, thanks.

Our command line build tools are implemented in XS. The tool shell defines [its host objects](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/tool.js) using the techniques above. The native implementation [clones a VM](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/tool.c#L59) to run the tools - it isn't much code.

creationix, 2018-08-01:
For more context, here is what I have now:
```c
#include <stdio.h>
#include <assert.h>
#include <stdbool.h>
#include <stdint.h>
#include <stdlib.h>

#include "xs.h"

///////////////////////////////////////////////////////////////////////////////
// Sample object hierarchy using union to share fields in inheritance.
///////////////////////////////////////////////////////////////////////////////

typedef struct Shape
{
  int x;
  int y;
} my_shape_t;

typedef struct Rectangle
{
  int x;
  int y;
  int w;
  int h;
} my_rectangle_t;

typedef struct Circle
{
  int x;
  int y;
  int r;
} my_circle_t;

// Union to force common layout for inheritance
union Shapes {
  my_shape_t shape;
  my_rectangle_t rect;
  my_circle_t circ;
};

///////////////////////////////////////////////////////////////////////////////
// Internal types, masks, and container
///////////////////////////////////////////////////////////////////////////////

// The concrete types that will have instances
typedef enum
{
  MyRectangle = 1 << 1,
  MyCircle = 1 << 2,
} my_type_t;

// All possible types that will have methods
typedef enum
{
  MyShapeMask = MyRectangle | MyCircle,
  MyRectangleMask = MyRectangle,
  MyCircleMask = MyCircle,
} my_mask_t;

// container that's allocated on the VM and managed by the GC
// We use a pointer for the actual object simulating an existing library that 
// doesn't let us own the memory.
typedef struct
{
  my_type_t type;
  void *ptr;
} my_container_t;

///////////////////////////////////////////////////////////////////////////////
// Helper to check type (including inheritance) and get pointer.
///////////////////////////////////////////////////////////////////////////////

static void *get_type(xsMachine *the, xsSlot slot, my_mask_t mask)
{
  my_container_t *c = xsGetHostChunk(slot);
  assert(mask & c->type); // TODO: throw JS TypeError instead of assert failing.
  return c->ptr;
}

///////////////////////////////////////////////////////////////////////////////
// Functions exposed to ECMAScript
///////////////////////////////////////////////////////////////////////////////

void my_greet(xsMachine *the)
{
  printf("Hello From greet\n");
}

void my_shape_position(xsMachine *the)
{
  my_shape_t *shape = get_type(the, xsThis, MyShapeMask);
  printf("Shape position x=%d y=%d\n", shape->x, shape->y);
}

void my_rectangle_area(xsMachine *the)
{
  my_rectangle_t *rectangle = get_type(the, xsThis, MyRectangleMask);
  int a = rectangle->x * rectangle->y;
  printf("Rectangle area w=%d * h=%d = a=%d\n", rectangle->x, rectangle->y, a);
}

void my_circle_area(xsMachine *the)
{
  my_circle_t *circle = get_type(the, xsThis, MyCircleMask);
  int a = 3141 * circle->r * circle->r / 1000;
  printf("Circle area PI * r=%d ^ 2 = a=%d\n", circle->r, a);
}

void my_shape_destructor(void *data)
{
  free(data);
}

void my_rectangle_constructor(xsMachine *the)
{
  my_rectangle_t *s = malloc(sizeof(*s));
  s->x = xsToInteger(xsArg(0));
  s->y = xsToInteger(xsArg(1));
  s->w = xsToInteger(xsArg(2));
  s->h = xsToInteger(xsArg(3));
  my_container_t c = {MyRectangle, s};
  xsSetHostChunk(xsThis, &c, sizeof(c));
}

void my_circle_constructor(xsMachine *the)
{
  my_circle_t *s = malloc(sizeof(*s));
  s->x = xsToInteger(xsArg(0));
  s->y = xsToInteger(xsArg(1));
  s->r = xsToInteger(xsArg(2));
  my_container_t c = {MyCircle, s};
  xsSetHostChunk(xsThis, &c, sizeof(c));
}

///////////////////////////////////////////////////////////////////////////////
// Main app testing design.
///////////////////////////////////////////////////////////////////////////////

int main(int argc, char *argv[])
{
  xsCreation creation = {
      128 * 1024 * 1024, /* initialChunkSize */
      16 * 1024 * 1024,  /* incrementalChunkSize */
      4 * 1024 * 1024,   /* initialHeapCount */
      1 * 1024 * 1024,   /* incrementalHeapCount */
      1024,              /* stackCount */
      2048 + 1024,       /* keyCount */
      1993,              /* nameModulo */
      127,               /* symbolModulo */
      32 * 1024,         /* parserBufferSize */
      1993,              /* parserTableModulo */
  };

  // Create a new machine
  xsMachine *the = xsCreateMachine(&creation, "myxs", NULL);
  assert(the);

  // Set a `global.greet` function.
  xsSet(xsGlobal, xsID("greet"), xsNewHostFunction(my_greet, 0));

  // Set up prototype chain and constructors.
  xsSlot objectProto = xsNewObject();
  xsSet(objectProto, xsID("position"), xsNewHostFunction(my_shape_position, 0));

  xsSlot circleProto = xsNewObject();
  xsSet(circleProto, xsID("__proto__"), objectProto);
  xsSet(circleProto, xsID("area"), xsNewHostFunction(my_circle_area, 0));
  xsSet(xsGlobal, xsID("Circle"), xsNewHostConstructor(my_circle_constructor, 3, circleProto));

  xsSlot rectangleProto = xsNewObject();
  xsSet(rectangleProto, xsID("__proto__"), objectProto);
  xsSet(rectangleProto, xsID("area"), xsNewHostFunction(my_rectangle_area, 0));
  xsSet(xsGlobal, xsID("Rectangle"), xsNewHostConstructor(my_rectangle_constructor, 4, rectangleProto));

  // TODO: hookup destructor somehow
  // We should probably swith to using `xsc` and `@` syntax files to connect binidngs.

  // Call `global.greet()`.
  xsCall0(xsGlobal, xsID("greet"));

  // Create an instance of Rectangle
  xsSlot rect = xsNew4(xsGlobal, xsID("Rectangle"), xsInteger(1), xsInteger(2), xsInteger(3), xsInteger(4));

  // Test rect.area()
  xsCall0(rect, xsID("area"));
  // Test rect.position() to test inheritance.
  xsCall0(rect, xsID("position"));

  xsDeleteMachine(the);
  return 0;
}
```

creationix, 2018-08-01:
Is there a way to implement the inheritance of host objects?

Also thanks for the links, that should help me construct the objects better.

creationix, 2018-08-01:
I guess for the inheritance, I could just manually flatten the methods and set the same C function for `position` on both prototypes.

phoddie, 2018-08-01:
> Is there a way to implement the inheritance of host objects?

Unfortunately, no. That's effectively multiple inheritance which JavaScript doesn't handle.

Setting aside the topic of constructing the objects, I gave the your code above a quick read. It looks reasonable. I'm suspicious of `my_shape_destructor` calling `free` as the data passed will be the host chunk, which is owned by the VM. If I understand the code correctly, you need to deference the chunk to get `ptr` and then free that.

You have a comment "// TODO: hookup destructor somehow".  `xsSetHostDestructor` will do that, though as already covered, better to let a class definition take care of that kind those details.

creationix, 2018-08-01:
>  I'm suspicious of my_shape_destructor calling free as the data passed will be the host chunk, which is owned by the VM. If I understand the code correctly, you need to deference the chunk to get ptr and then free that.

I misunderstood the docs then.  I thought the `void*` passed in would be my `ptr` field.  Thanks for the catch.

creationix, 2018-08-01:
So in my new version, I've got the following `shapes.js` module:

```js
export class Circle @ "my_shape_destructor" {
    constructor(x, y, r) @ "my_circle_constructor"
    position() @ "my_shape_position"
    area() @ "my_circle_position"
}

export class Rectangle @ "my_shape_destructor" {
    constructor(x, y, r) @ "my_rectangle_constructor"
    position() @ "my_shape_position"
    area() @ "my_rectangle_position"
}
```

I know I need to compile this with `msc -c ...`.  I'm just not sure what options are best and how to integrate with my Makefile.

creationix, 2018-08-01:
Also I have the following test script:
```js
import { Circle, Rectangle } from "shapes"

let c = new Circle(30, 20, 10)
c.area()
c.position()

let r = new Rectangle(10, 20, 30, 40)
r.area()
r.position()
```

This represents code my users will be writing.  Since I'm not on a super constrained microcontroller, I don't have to pre-compile this script and can include it as a resource and use eval.

I'm not sure which will be a better workflow for my users.  There is also a packaging step in my SDK, so I could probably run `xsc` on their code as well.  What I'm unsure of is if the linking has to be done all at once (for my internal bindings and for their user code).  I'd rather skip C compiling on their end if possible.

creationix, 2018-08-02:
> It is possible to run xsc to, generate an xs binary file (XSB) and load that into an existing VM. But, it is tricky and rare.

Is this because the app might use symbols not in the base image?

Would it be better to use eval on the device?  I saw a posting somewhere saying that is an option now.

phoddie, 2018-08-02:
> Is this because the user-code might use symbols not in the base image?

It is OK for user-code to use additional keys. We try to avoid that, but it works. If you expect user code to do that a lot, then the key table needs to be configured (using `the->keyCount`) with room for more runtime defined keys.

Much of the complexity of loading a pre-compiled user binary on the device has to do with the finding the module. That's entirely host defined (e.g. you have to decide that as there's no definition in the standard). The module then needs to be remapped so its key (symbol) ids match those of the VM. XS has code to do that for you, but it adds the requirement that the loaded module be read/write, at least at load time, which is tricky on device where we want the module to reside in ROM.

> Would it be better to use eval on the device? I saw a posting somewhere saying that is an option now.

`eval` on the device is supported but really only recommend for things like a command line / REPL or for testing (the test252 suite doesn't work at all without `eval`). Here's the blog post introducing support for on-device parsing: http://blog.moddable.com/blog/eval/.

creationix, 2018-08-02:
When I define my hybrid classes with your suggested JS, the subclasses don't have a host object in the constructor:

```js
class Shape @ "my_shape_destructor" {
    position() @ "my_shape_position"
}

export class Circle extends Shape {
    constructor(x, y, r) @ "my_circle_constructor"
    area() @ "my_circle_position"
}

export class Rectangle extends Shape {
    constructor(x, y, w, h) @ "my_rectangle_constructor"
    area() @ "my_rectangle_position"
}
```

If I flatten it manually, then it works, but it's ugly and will cause trouble when I use large object hierarchy trees.

```js
export class Circle @ "my_shape_destructor" {
    constructor(x, y, r) @ "my_circle_constructor"
    position() @ "my_shape_position"
    area() @ "my_circle_area"
}

export class Rectangle @ "my_shape_destructor" {
    constructor(x, y, w, h) @ "my_rectangle_constructor"
    position() @ "my_shape_position"
    area() @ "my_rectangle_area"
}
```

Is there a way to call `xsSetHostChunk` on objects who inherit from a host object type?

phoddie, 2018-08-02:
Right, there's no host data yet so it throws. Someone has to create the host data (e.g. call `xsSetHostChunk`). The VM doesn't know enough to do that itself. It is fine for you to create it from the subclass's constructor (as long as that is what the super class expects)..There's only one host data on the object - not a separate one for each level of inheritance.

creationix, 2018-08-02:
I'm calling `xsSetHostChunk` in the subclass' [constructor](https://github.com/creationix/xs-shapes/blob/master/src/shapes.c#L57) and that's where it's throwing the exception.

creationix, 2018-08-02:
Debugging it, my `xsThis` slot has `it->kind` value of `8` (where `XS_REFERENCE_KIND` is `9`) so it fails the `fxCheckHostObject` call and throws the exception.

creationix, 2018-08-02:
Hmm, doesn't seem to be working for me.  See my latest code https://github.com/creationix/xs-shapes/blob/dd7677eb382185666c434e31d181515efa288bcd/src/shapes.js#L3-L5

Also, possibly unrelated, but my destructor gets called lots with null pointers passed in.

```
### SyntaxError: Circle: C: xsSetHostData: no host object
(nil): my_shape_destructor
(nil): my_shape_destructor
(nil): my_shape_destructor
```

phoddie, 2018-08-02:
You have to change the `constructor` property of the `Circle` and `Rectangle` classes to `construct`. See that above. The way it works is that Shape's constructor is always called first. That completes initialization and then Shape's constructor calls `construct` on the subclass.

The destructor of native objects is called with NULL when you terminate the virtual machine. You see three calls because you have three classes sharing a destructor. The usual pattern in a destructor is to first check for data being NULL. That also guards against exceptions in the constructor being calling `xsSetHostData` or `xsSetHostChunk`.

creationix, 2018-08-02:
Thanks!  Now I'm making progress...

Just need to implement a `print` global and I think this example is done.

creationix, 2018-08-03:
Thanks for the help. You can close this if you want or leave it open so others can find it easier.

closed 2018-08-03 by phoddie