Moddable-OpenSource/moddable #77: Possible bugs detected by Infer

creationix, 2018-08-03:
[Infer](http://fbinfer.com/) is a static analysis tool written by Facebook that I'm trying on my code base.  I found a lot of warnings in xs itself and so decided to report the warnings detected by compiling just `xsc`.

```
Found 45 issues

xs/sources/xsCode.c:206: error: UNINITIALIZED_VALUE
  The value read from coder.firstCode was never initialized.
  204.   	c_memset(script, 0, sizeof(txScript));
  205.   	
  206. > 	code = coder.firstCode;
  207.   	size = 0;
  208.   	delta = 0;

xs/sources/xsCode.c:339: error: UNINITIALIZED_VALUE
  The value read from coder.firstCode was never initialized.
  337.   	}	
  338.   	
  339. > 	code = coder.firstCode;
  340.   	size = 0;
  341.   	while (code) {

xs/sources/xsCode.c:512: error: UNINITIALIZED_VALUE
  The value read from coder.firstCode was never initialized.
  510.   	script->codeSize = size;
  511.   	
  512. > 	code = coder.firstCode;
  513.   	p = script->codeBuffer;
  514.   	while (code) {

xs/tools/xsc.c:220: error: UNINITIALIZED_VALUE
  The value read from c was never initialized.
  218.   	fprintf(file, "void xsHostModule(xsMachine* the)\n");
  219.   	fprintf(file, "{\n");
  220. > 	fprintf(file, "\tstatic xsHostBuilder builders[%d] = {\n", c);
  221.   	for (i = 0; i < c; i++) {
  222.   		txS1 length = *p++;

xs/tools/xsc.c:231: error: UNINITIALIZED_VALUE
  The value read from c was never initialized.
  229.   	}
  230.   	fprintf(file, "\t};\n");
  231. > 	fprintf(file, "\txsResult = xsBuildHosts(%d, builders);\n", c);
  232.   	fprintf(file, "}\n\n");
  233.   }

xs/tools/xsc.c:227: error: UNINITIALIZED_VALUE
  The value read from id was never initialized.
  225.   			fprintf(file, "\t\t{ (xsCallback)%s, -1, -1 },\n", p);
  226.   		else
  227. > 			fprintf(file, "\t\t{ %s, %d, %d },\n", p, length, id);
  228.   		p += c_strlen((char*)p) + 1;
  229.   	}

xs/tools/xsc.c:416: error: UNINITIALIZED_VALUE
  The value read from length was never initialized.
  414.   			mxParserThrowElse(fwrite(&size, 4, 1, file) == 1);
  415.   			mxParserThrowElse(fwrite("NAME", 4, 1, file) == 1);
  416. > 			mxParserThrowElse(fwrite(rename, length, 1, file) == 1);
  417.   		}
  418.   

xs/sources/xsLexical.c:314: error: DEAD_STORE
  The value written to &p (type char*) is never used.
  312.   			fxReportParserError(parser, "invalid number");			
  313.   	}
  314. > 	*p++ = 0;
  315.   	fxGetNextNumber(parser, fxStringToNumber(parser->dtoa, parser->buffer, 1));
  316.   }

xs/sources/xsre.c:470: error: DEAD_STORE
  The value written to &current0 (type int*) is never used.
  468.   		*current0++ = character;
  469.   		count0++;
  470. > 		*current0++ = 0x7FFFFFFF;
  471.   	}
  472.   	result->characters[0] = count0;

xs/sources/xsre.c:503: error: UNINITIALIZED_VALUE
  The value read from character was never initialized.
  501.   		if ((flag == 0) || (old == 0)) {
  502.   			count0++;
  503. > 			*current0++ = character;
  504.   		}
  505.   		old = flag;

...too many issues to display (limit=10 exceeded), please see /home/tcaswell/Code/xs-shapes/moddable/infer-out/bugs.txt or run `infer-explore` for the remaining issues.

Summary of the reports

           DEAD_STORE: 30
  UNINITIALIZED_VALUE: 13
     NULL_DEREFERENCE: 1
          MEMORY_LEAK: 1
```

Hopefully this is helpful.  If you'd like I might find time to submit some PRs to moddable itself if you're open to such contributions?

creationix, 2018-08-03:
Here is the full list in abbreviated form:

```
 0. xs/sources/xsCode.c:206: error: UNINITIALIZED_VALUE
      The value read from coder.firstCode was never initialized.

 1. xs/sources/xsCode.c:339: error: UNINITIALIZED_VALUE
      The value read from coder.firstCode was never initialized.

 2. xs/sources/xsCode.c:512: error: UNINITIALIZED_VALUE
      The value read from coder.firstCode was never initialized.

 3. xs/tools/xsc.c:220: error: UNINITIALIZED_VALUE
      The value read from c was never initialized.

 4. xs/tools/xsc.c:231: error: UNINITIALIZED_VALUE
      The value read from c was never initialized.

 5. xs/tools/xsc.c:227: error: UNINITIALIZED_VALUE
      The value read from id was never initialized.

 6. xs/tools/xsc.c:416: error: UNINITIALIZED_VALUE
      The value read from length was never initialized.

 7. xs/sources/xsLexical.c:314: error: DEAD_STORE
      The value written to &p (type char*) is never used.

 8. xs/sources/xsre.c:470: error: DEAD_STORE
      The value written to &current0 (type int*) is never used.

 9. xs/sources/xsre.c:503: error: UNINITIALIZED_VALUE
      The value read from character was never initialized.

10. xs/sources/xsLexical.c:1057: error: DEAD_STORE
      The value written to &q (type char*) is never used.

11. xs/sources/xsScope.c:979: error: DEAD_STORE
      The value written to &scopeLevel (type int) is never used.

12. xs/sources/xsScope.c:980: error: DEAD_STORE
      The value written to &scopeMaximum (type int) is never used.

13. xs/sources/xsre.c:1146: error: NULL_DEREFERENCE
      pointer `term` last assigned on line 1143 could be null and is dereferenced at line 1146, column 2.

14. xs/sources/xsre.c:1297: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

15. xs/sources/xsre.c:1302: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

16. xs/sources/xsre.c:1308: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

17. xs/sources/xsre.c:1313: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

18. xs/sources/xsre.c:1326: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

19. xs/sources/xsre.c:1334: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

20. xs/sources/xsre.c:1352: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

21. xs/sources/xsre.c:1378: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

22. xs/sources/xsre.c:1388: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

23. xs/sources/xsre.c:1396: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

24. xs/sources/xsre.c:1404: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

25. xs/sources/xsre.c:1415: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

26. xs/sources/xsre.c:1425: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

27. xs/sources/xsre.c:1431: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

28. xs/sources/xsre.c:1452: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

29. xs/sources/xsre.c:1460: error: DEAD_STORE
      The value written to &buffer (type int*) is never used.

30. xs/sources/xsLexical.c:1650: error: DEAD_STORE
      The value written to &q (type char*) is never used.

31. xs/sources/xsCode.c:1697: error: UNINITIALIZED_VALUE
      The value read from value was never initialized.

32. xs/sources/xsre.c:1763: error: MEMORY_LEAK
      `state` is not reachable after line 1763, column 8.

33. xs/sources/xsre.c:1862: error: DEAD_STORE
      The value written to &pointer (type int*) is never used.

34. xs/sources/xsre.c:1850: error: DEAD_STORE
      The value written to &quantifiers (type txQuantifierData*) is never used.

35. xs/sources/xsdtoa.c:2277: error: UNINITIALIZED_VALUE
      The value read from u.d was never initialized.

36. xs/sources/xsdtoa.c:2304: error: DEAD_STORE
      The value written to &xa (type unsigned int*) is never used.

37. xs/sources/xsdtoa.c:2311: error: DEAD_STORE
      The value written to &xa (type unsigned int*) is never used.

38. xs/sources/xsdtoa.c:2342: error: UNINITIALIZED_VALUE
      The value read from d.d was never initialized.

39. xs/sources/xsSyntaxical.c:2900: error: DEAD_STORE
      The value written to &binding (type sxNode*) is never used.

40. xs/sources/xsCode.c:3006: error: UNINITIALIZED_VALUE
      The value read from value was never initialized.

41. xs/sources/xsCode.c:3076: error: UNINITIALIZED_VALUE
      The value read from value was never initialized.

42. xs/sources/xsdtoa.c:3323: error: DEAD_STORE
      The value written to &bbits (type int) is never used.

43. xs/sources/xsdtoa.c:4397: error: DEAD_STORE
      The value written to &Lsb (type unsigned int) is never used.

44. xs/sources/xsdtoa.c:5372: error: DEAD_STORE
      The value written to &blen (type unsigned long) is never used.
```

creationix, 2018-08-03:
It looks like a lot of the `UNINITIALIZED_VALUE` warnings are because you alias `memset` to `c_memset`.  I'm not sure if this is helpful after all...

phoddie, 2018-08-03:
Thanks. The results of static analyzers can be useful. For some time we ran against Coverity. More recently we run against the Clang clang static analyzer. They find some real issues, but they also get tripped on code in the interpreter that is correct (as far as we can tell!) but isn't a common pattern.

FWIW - we try to alias all the standard C functions through a `c_` prefix so that it is easy and safe to redirect them to different implementations. That is sometimes useful on a desktop or mobile, but absolutely essential for microcontrollers which may not even include all the standard functions (or have limited / buggy implementation, etc).

If you see issues that look real or have fixes, we'd be interested, of course.

creationix, 2018-08-03:
Yep, I just ran the clang analyzer on the code and the uninitialized value warnings went away.  I guess the fb tool isn't mature enough yet.  Sorry for the noise.

closed 2018-08-03 by creationix