Moddable-OpenSource/moddable #79: Is it possible to link with `xsl`'s output using only public APIs?

creationix, 2018-08-03:
From what I can tell, the `xsAll.h` header is meant for internal use and brings in all kinds of `sx.` and `fx...` namespacing whereas the `xs...` symbols in `xs.h` and `xsmc...` in `xsmc.h` looks public.

I'm having trouble including `xsAll.h` in my `main.cpp` file.  I get all kinds of compiler errors, it looks like it brings in code that's not valid C++.

I'd love to stick to just `xs.h` or `xsmc.h`, but I can't seem to figure out how to hook up the `xsl` output since it uses the `txPreparation* xsPreparation()` function in `mc.xs.c` (which isn't even in the `mc.xs.h` so I'm thinking this is a private API as well)

Please tell me there is a way to keep `xsc` and `xsl` in my workflow, but avoid `xsAll.h` and `extern txPreparation* xsPreparation()`.

phoddie, 2018-08-03:
Right - `xs.h` and `xsmc.h` are the public interfaces. You shouldn't need to include `xsAll.h` unless you are working on the internals. The headers are definitely C, not C++, and wrapped in `extern "C" {`.

Rather than try to sort out why `xsAll.h` is failing for you, let's look at moving the `xsPreparation` related files to `xs.h`. That will take a little investigation and experimentation to avoid breaking our builds.

creationix, 2018-08-06:
Btw, the issue I'm having with the internal code is the following struct:
```
typedef struct {
	txTypeCallback add;
	txTypeCallback and;
	txTypeCallback compareExchange;
	txTypeCallback exchange;
	txTypeCallback load;
	txTypeCallback or;
	txTypeCallback store;
	txTypeCallback sub;
	txTypeCallback xor;
} txTypeAtomics;
```

This is a syntax error in many versions of C because `and`, `or`, and `xor` are not valid identifiers.

creationix, 2018-08-06:
If I rename them in this header and `xsAtomics.c` to be `and_`, `or_` and `xor_`, it gets farther than then fails on:
```c
typedef union {
...
	struct { txSlot* closure; txSlot* module; } export;
} txValue;
```

I assume because `export` is a reserved word.

phoddie, 2018-08-06:
> Any idea where I can start with moving the xsPreparation stuff to the public API? 

We've got a solution to the preparation stuff. That'll get pushed in a day or so.

> This is a syntax error in many versions of C because and, or, and xor are not valid identifiers

Wacky. Will give a look.

creationix, 2018-08-06:
> We've got a solution to the preparation stuff. That'll get pushed in a day or so.

Awesome, I've eagerly await.

> Wacky. Will give a look.

FYI, my actual application is C++ based (I'm fairly new to C++), so maybe it's only a syntax error when compiling as C++ (C isn't really a strict subset I'm learning).  Obviously it worked fine in my public sample where my code was C99 and GNU99.

phoddie, 2018-08-09:
Yes, it is now possible. Use `xsPrepareMachine()` in place of `xsCloneMachine()`. There are a few examples in the repository now. Check out [tool.c](https://github.com/Moddable-OpenSource/moddable/blob/public/tools/tool.c#L61) for an example.

closed 2018-08-09 by phoddie