Moddable-OpenSource/moddable #80: Generated C code doesn't compile on MacOS clang.

creationix, 2018-08-10:
I've almost got XS working in all my test environments.  The last issue it seems is the style of code generated for `mc.xs.c`.  When compiling using clang, I get:

```
/Users/tcaswell/Code/xs-shapes/out/mc.xs.c:125:87: error: designator in initializer for scalar type 'txInteger' (aka 'int')
/* 0000 */      { NULL, {.ID = XS_NO_ID, .flag = XS_NO_FLAG, .kind = XS_REFERENCE_KIND}, { .reference = (txSlot*)&gxHeap[3352] } },
                                                                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/tcaswell/Code/xs-shapes/out/mc.xs.c:126:73: error: designator in initializer for scalar type 'txInteger' (aka 'int')
/* 0001 */      { NULL, {.ID = 0, .flag = 0xc0, .kind = XS_INSTANCE_KIND}, { .instance = { NULL, NULL } } },
                                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
/Users/tcaswell/Code/xs-shapes/out/mc.xs.c:127:73: error: designator in initializer for scalar type 'txInteger' (aka 'int')
/* 0002 */      { NULL, {.ID = 1, .flag = 0xc0, .kind = XS_INSTANCE_KIND}, { .instance = { NULL, NULL } } },
                                                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~
...
```

I know I've seen this on newer clang versions (this particular dump is `Apple LLVM version 9.1.0 (clang-902.0.39.2)`) .

creationix, 2018-08-10:
Looking closer, I see there is an `if (dummy) fprintf(file, "0, ");` in each of the places I added the `.value =` . I guess my platform isn't setting this `dummy` flag when it should.

Would it not be better to just use the designators and let the compiler match up fields than have dynamically generated dummy fields?

creationix, 2018-08-10:
Here is how I worked around it in my branch: https://github.com/creationix/moddable/commit/d586091663b9bbda3252cf29f279b19a878e0ac3

phoddie, 2018-08-10:
Interesting. I'm at Clang 9.0 on macOS (skipped an update at some point) and don't see the warning. Your fix looks reasonable.

phoddie, 2018-08-18:
We recently committed a solution to this that, as you proposed, avoids the "dummy" entirely. It also eliminates the "-t" option from xsl as the same generated code now works on 32 and 64-bit targets.

closed 2018-08-18 by phoddie