Moddable-OpenSource/moddable #82: XS: For-in prints deleted keys

bakkot, 2018-08-27:
(Sorry if this is the wrong repo; I'm not sure where to file bugs in the JavaScript engine.)

Consider the following program:

```js
if (typeof console === 'undefined') console = { log: print };

let a = {
  x: 0,
  y: 1,
};

for (let key in a) {
  console.log(key);
  delete a.y;
  if (key === 'y') {
    console.log('reached');
  }
}
```

This prints `x, y, reached`. It should only print `x`. It should not reach the step with `y`, since `y` has been deleted before it gets to that iteration of the loop. The spec is pretty loose here, but does not allow this behavior: in #sec-enumerate-object-properties, it says "A property that is deleted before it is processed by the iterator's next method is ignored." No other engine has this behavior, though see related [spidermonkey bug](https://bugzilla.mozilla.org/show_bug.cgi?id=1486656) - spidermonkey exhibits this behavior when iterating over proxy objects, even those which are totally transparent.

See also (and please comment on) this open spec bug about more precisely specifying the behavior of for-in, which prompted the investigation which lead me to discovering this: https://github.com/tc39/ecma262/issues/1281

PS: @patrick-soquet, thanks for [adding](https://github.com/GoogleChromeLabs/jsvu/pull/22) moddable support to eshost! It makes it much, much more likely I'll find and file bugs like this.

phoddie, 2018-08-27:
@bakkot, this is exactly the right place to report issues like this one. Thank you.

patrick-soquet, 2018-08-28:
Thank you for the bug report. Fix committed, it will be in the open source tree soon.

I also published a new version of `xst` that should be available in `eshost` for you to try.

Regarding [tc39/ecma262#1281](https://github.com/tc39/ecma262/issues/1281), XS now prints the same results as the informative implementation given in the specification:

 **1**
```
a getOwnPropertyDescriptor: 0
0
a getOwnPropertyDescriptor: 1
1
b getOwnPropertyDescriptor: 1
b getOwnPropertyDescriptor: 2
2
```
 **2**
```
a getOwnPropertyDescriptor: 0
0
a getOwnPropertyDescriptor: 1
b getOwnPropertyDescriptor: 1
1
b getOwnPropertyDescriptor: 2
2
```
 **3**
```
a getOwnPropertyDescriptor: 0
0
a getOwnPropertyDescriptor: 1
1
b getOwnPropertyDescriptor: 1
```

phoddie, 2018-08-29:
Fix committed.

closed 2018-08-29 by phoddie