Moddable-OpenSource/moddable #83: AudioOut to DAC pin

diogoviannaaraujo, 2018-09-02:
Hi,

Ive been trying to do some audio out with ESP32, I've seen your i2s/audioout module but im having some questions.

Is it possible to use it to output audio to one of the DAC pins? Ive been reading all your code and I could find a MODDEF_AUDIOOUT_I2S_DAC define, but I see no documentation of if.

And is there any manifest or ready example using audioout? I found some JS code using It on piu/mc but there is no manifest

Thanks In advance

diogoviannaaraujo, 2018-09-03:
Hi, I've found out that defining the MODDEF_AUDIOOUT_I2S_DAC as 1 will use the DAC as the output for the i2s. Ive been able to reproduce a sound file using only a simple amplifier.

As soon I have more time will contribute some exemples on pins/i2s and some documentation on wav2maud too

phoddie, 2018-09-03:
@diogoviannaaraujo - Thanks for looking at this. The audio implementation for ESP32 and ESP works for many cases, but it hasn't been tested thoroughly. That's why there isn't documentation and examples yet. If you have time to help fill that in, it would be great.

diogoviannaaraujo, 2018-09-10:
I guess this is already answered. In the future I guess we can discuss the documentation on a pull request.

closed 2018-09-10 by diogoviannaaraujo