Moddable-OpenSource/moddable #84: Random Build Errors on MacOS

diogoviannaaraujo, 2018-09-06:
Hi,

When building an example on MacOS usually the first time fails.
There are a lot of messages I dont understand but I think the errors are shown in this part:

```
arser.o
AR /Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/idf/ethernet/libethernet.a
make[2]: echo: Resource temporarily unavailable
AR /Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/idf/heap/libheap.a
AR /Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/idf/openssl/libopenssl.a
Building partitions from /Users/diogoviannaaraujo/Git/moddable/build/devices/esp32/xsProj/partitions.csv...
make[2]: *** [src/fast_crypto/fast_crypto_internal-cipher.o] Error 1
make[2]: *** Waiting for unfinished jobs....
AR /Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/idf/log/liblog.a
make[2]: *** [wifi_init.o] Error 1
make[2]: *** Waiting for unfinished jobs....
AR /Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/idf/coap/libcoap.a
CC /Users/diogoviannaaraujo/Git/moddable/build/tmp/esp32/debug/idf/mbedtls/mbedtls/library/cipher_wrap.o
idf/mbedtls/mbedtls/library/chachapoly.o
make[2]: xtensa-esp32-elf-gcc: Resource temporarily unavailable
make[2]: *** [src/diskio_sdmmc.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [mbedtls/library/padlock.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [core/mem.o] Error 1
make[2]: *** Waiting for unfinished jobs....
```

phoddie, 2018-09-06:
This happens because the ESP32 IDF build isn't set-up to safely run parallel build jobs. It doesn't happen consistently, and a second build is usually enough. It is annoying. There's an option in our build to build using a single job, which always works but is slower. We should document that (I don't remember right now myself...).

bfriedkin, 2018-09-06:
Diogo -

The ESP32 IDF build runs parallel build jobs, but sometimes that ends up consuming too many system resources.

You can cap the number of jobs run by setting the MAKEFLAGS_JOBS environment variable:

export MAKEFLAGS_JOBS="-j 4"

I build on macOS using the setting above and found that the ESP32 build runs much faster. Plus the macOS host is more responsive during the build since less resources are used.

Regards,
Brian

diogoviannaaraujo, 2018-09-10:
Since this a configuration with IDF and not something from Moddable I`ll be closing this issue

closed 2018-09-10 by diogoviannaaraujo