Moddable-OpenSource/moddable #85: modLog a variable

diogoviannaaraujo, 2018-09-10:
Hi, I've been trying to use modLog to print a variable. Is it possible?
Ive tried to pass like I would do in printf or in ESP_LOGI but it takes only one parameter.
Also tried to pass a variable directly but I get something about wrong initialization.
Is my variable in wrong type?
I bet is a miss understanding about how C work but Im scratching my head on this..

phoddie, 2018-09-11:
In a debug build, the output of modLog is routed to xsbug. You'll see it in the console log there.

> `modLog` is intended for application level use. If you call it from an interrupt, it may corrupt the log in xsbug.

For a release build, output goes to UART 0. For that to work, you need to enable console output in SDKCONFIG.release:

`````
CONFIG_CONSOLE_UART_DEFAULT=y
CONFIG_CONSOLE_UART_CUSTOM=
CONFIG_CONSOLE_UART_NONE=
CONFIG_CONSOLE_UART_NUM=0
CONFIG_CONSOLE_UART_BAUDRATE=115200
`````

phoddie, 2018-10-19:
A few more notes, modLog is a very simple tool to avoid requiring the overhead of printf (which is not always available on the host platforms). Here is what is available:

- modLog("foo") - this must be a constant string. On platforms like ESP8266 which put strings in RAM by default, this allows the strings to be placed in ROM.
- modLogVar(string) - this can be a string variable (e.g .char *)
- modLogInt(integer) - 32-bit integer, decimal output
- modLogHex(integer) - 32-bit integer, hexadecimal output

closed 2018-10-19 by phoddie