Moddable-OpenSource/moddable #86: Generating a .bin

diogoviannaaraujo, 2018-09-10:
does mcconfig has any parameter to just generate a .bin to cashing with other flashing tool?

mkellner, 2018-09-10:
in tools/mcconfig/make.esp32.mk is the main piece of the makefile that mcconfig uses to build the esp32 bits.

Look for this section in the “all:” target:

    cd $(PROJ_DIR) ; IDF_BUILD_DIR=$(IDF_BUILD_DIR) DEBUG=$(DEBUG) SDKCONFIG_DEFAULTS=$(SDKCONFIG_FILE) make flash;
    $(DO_LAUNCH)

If you change the “make flash” to just “make” and remove the $(DO_LAUNCH), then mcconfig will just build the file.

The .bin (xs_esp32.bin)file and friends (partitions.bin, bootloader.bin, etc.) will be in moddable/build/tmp/esp32/debug/idf/

diogoviannaaraujo, 2018-09-11:
Ive done a normal upload to a board and then dumped the bin with read_flash, had not the time to test if that worked and sended it that way.. Hope it will work

closed 2018-10-19 by Moddable-OpenSource