Moddable-OpenSource/moddable #87: Numeric keys larger than 2**24 - 2*20 + 1 are given as the empty string

bakkot, 2018-09-12:
Consider the following program:

```js
let obj = {
  [2**24 - 2**20]: 0,
};

let obj2 = {
  [2**24 - 2**20 + 1]: 0,
};

function isEmptyKey(thing) {
  let keys = Reflect.ownKeys(thing);
  if (keys.length !== 1) throw 'too many keys';
  console.log(keys[0] === '');
}

isEmptyKey(obj);
isEmptyKey(obj2);
```

This prints `false, true`. In ever other engine, and per [the spec](https://tc39.github.io/ecma262/#sec-string-exotic-objects-ownpropertykeys), it should be `false, false`. (No one ever hits the error path; that's just for illustration.)

phoddie, 2018-09-12:
That seems very obscure! Thanks for the report. We'll take a look.

phoddie, 2018-09-12:
Patrick has a fix for this one. We'll push that to GitHub in the next couple days.

closed 2018-10-01 by phoddie