Moddable-OpenSource/moddable #88: mcconfig : command not found error

SharmisthaSG, 2018-09-13:
I'm using Ubuntu 16.04. I followed all the steps for building and configuring Moddable SDK for ESP32. However, at the final step, I get command not found error for mcconfig.

mkellner, 2018-09-13:
Hello. Did you add $MODDABLE/build/bin/lin/release to your PATH?

SharmisthaSG, 2018-09-14:
Hey
@mkellner I tried both your suggestions. It still says command not found. I noticed an error in the previous step (make install) which I had somehow overlooked. Attaching a screenshot of the same. Maybe this can help.

[image]

@Frijol I did check the directory. I was in the correct one.

SharmisthaSG, 2018-09-14:
realpath ../.. doesn't return any path. And BUILD_DIR isn't set either. How do I resolve this?

mkellner, 2018-09-14:
Do you have `realpath`? (ie. what does `which realpath` return?)
If you don't have realpath, you can install it.
https://unix.stackexchange.com/questions/101080/realpath-command-not-found
Suggests how to install.
`$ sudo apt-get install realpath`

closed 2018-10-19 by Moddable-OpenSource