Moddable-OpenSource/moddable #89: Filepath assumptions in getting started

Frijol, 2018-09-13:
Hi team! Enjoying getting started with Moddable.

In your MacOS setup (https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20%E2%80%93%20Getting%20Started.md), you have an assumption built into your path instructions:

[image]

It assumes you're `git clone`ing into a folder called Projects, but doesn't explicitly say you should. Might want to make this assumption more explicit to avoid confusion on why the MODDABLE path doesn't work!

bfriedkin, 2018-09-13:
Hello -

Thank you for the suggestion! We will take a look and try to improve the instructions.

Regards,
Brian

bfriedkin, 2018-10-20:
Hello again -

We have updated the Getting Started instructions for all host platforms to be more explicit about the Projects folder. Thank you again for pointing that out. The documentation update will be available in the next open source release.

Regards,
Brian

closed 2018-10-20 by bfriedkin