Moddable-OpenSource/moddable #92: Esp32 ble close function wrongly documented

diogoviannaaraujo, 2018-09-30:
Hi, in ble documentation

bfriedkin, 2018-10-01:
Diogo -

Thank you for reporting this issue. The close() call does terminate any client connection, but also releases all BLE resources.

We will add a disconnect() call that disconnects the client but leaves the server running and also update the documentation accordingly.

Regards,
Brian

diogoviannaaraujo, 2018-10-01:
I’ll have some time later in the week so I can play a little bit and see what I can do.

I think the most important is to define who to disconnect so it won’t disconnect a already authorized client.

bfriedkin, 2018-10-19:
Diogo -

We have added a disconnect() call to the [BLEServer class](
https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/ble/bleserver.js) and updated the documentation. The updates will be available in our next Open Source release.

Regards,
Brian

closed 2018-10-19 by bfriedkin