Moddable-OpenSource/moddable #93: Windows Getting Started fails

AnthonyKNorman, 2018-10-12:
Issue #1 was the assumptions re downloading into 'Project' folder
Issue#2 was could not find nmake - added path "C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\Hostx86\x86" into PATH
issue #3 

C:\Users\antho\moddable\build\makefiles\win>build

C:\Users\antho\moddable\build\makefiles\win>nmake GOAL=debug BUILD_DIR=C:\Users\antho\moddable\build XS_DIR=C:\Users\antho\moddable\xs /c /f C:\Users\antho\moddable\xs\makefiles\win\xsc.mak /s
xsdtoa.c
C:\Users\antho\moddable\xs\platforms\xsPlatform.h(85): fatal error C1083: Cannot open include file: 'ctype.h': No such file or directory
NMAKE :  U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2017\Community\VC\Tools\MSVC\14.15.26726\bin\Hostx64\x64\cl.EXE"' : return code '0x2'
Stop.

Should ctype.h be in the Git download?

bfriedkin, 2018-10-12:
Hello -

Thank you for the follow-up. Looking at the error, I am wondering if you are building from the "Developer Command Prompt for VS 2017" command line console. This is documented in the [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20–%20Getting%20Started.md) document in step 5. This command line console sets up additional environment variables when opened, including the path to the toolchain required by the Moddable SDK build. You shouldn't need to explicitly add the path to nmake or any other Visual Studio tools.

Please let us know if building from the "Developer Command Prompt for VS 2017" command line console resolves the build issue.

Thanks,
Brian

AnthonyKNorman, 2018-10-13:
Hi
D'Oh! Thanks for that. Rule 1 -  read ALL the words!
I now find that the "Developer Command Prompt for VS 2017" command line console does not recognise the MODDABLE environment variable
```
'%MODDABLE%' is not recognized as an internal or external command,
operable program or batch file.

```

In a Windows command prompt this works fine

```
C:\Users\antho\moddable\build\makefiles\win>cd %MODDABLE%
C:\Users\antho\moddable>
```

What am I doing wrong

bfriedkin, 2018-10-13:
Hello -

Let's first make sure that the tools build completed successfully. Navigate to %MODDABLE%\build\bin\win\release and verify that xsbug.exe and mcconfig.bat are in that directory.

If these files exist, then I would suggest editing your %MODDABLE% PATH environment variable entry to exclude the %MODDABLE% variable. When I print my PATH, my %MODDABLE% path entry is a full path without any environment variable. The way the PATH environment variable is interpreted and defined may vary depending on your version of Windows.

So in your PATH environment variable definition, change this:

%MODDABLE%\build\bin\win\release

To this:

C:\Users\antho\moddable\build\win\release

Thanks,
Brian

AnthonyKNorman, 2018-10-13:
OK, Using the full path in the PATH environment variable definition did the trick.
We have "Hello, world - sample" in the console!
Thanks for all your help. Going to hook up my ESP32 now!
Tony

bfriedkin, 2018-10-15:
Hello -

It looks like Espressif updated their all-in-one toolchain for Windows as well as the ESP-IDF, since we last updated our open source repository. We are working through updating our build. 

Thank you for your patience. I will write back here after we've posted an update.

Thanks,
Brian

bfriedkin, 2018-10-19:
Hello -

We have updated our build to be compatible with the recent Windows all-in-one toolchain and ESP-IDF updates from Espressif. The fixes will be available in our next Open Source release. Please follow the updated Windows setup instructions in our Getting Started document.

Regards,
Brian

closed 2018-10-19 by bfriedkin