Moddable-OpenSource/moddable #101: Problem flashing esp32/moddable_zero

AnthonyKNorman, 2018-11-04:
If I use
```bash
cd %MODDABLE%\examples\helloworld
mcconfig -d -m -p esp32
```
Then the ESP32 is flashed on COM3 and the example runs

If I use
```bash
cd %MODDABLE%\examples\piu\balls
mcconfig -d -m -p esp32/moddable_zero
```
then the MinGW window shows
```
Building xs_esp32.elf...
CC /c/Users/antho/moddable/build/tmp/esp32/moddable_zero/debug/idf/main/main.o
AR /c/Users/antho/moddable/build/tmp/esp32/moddable_zero/debug/idf/main/libmain.a
LD C:/Users/antho/moddable/build/tmp/esp32/moddable_zero/debug/idf/xs_esp32.elf
esptool.py v2.6-beta1
Flashing binaries to serial port COM8 (app at offset 0x10000 )...
esptool.py v2.6-beta1
Serial port COM8
```
Running SET in  "Developer Command Prompt for VS 2017" 
gives
```
UniversalCRTSdkDir=C:\Program Files (x86)\Windows Kits\10\
UPLOAD_PORT=COM3
USERDOMAIN=DESKTOP-G6SVJA3
```

Where is it getting COM8 from?

mkellner, 2018-11-04:
My guess is that at some point when you first built this app, you had the UPLOAD_PORT set to COM8? Then the sdkconfig.h file would be generated.

Try building clean (ie. remove $MODDABLE/build/bin/esp32 and $MODDABLE/build/tmp/esp32 ), or maybe just touch $MODDABLE/build/devices/esp32/xsProj/sdkconfig.defaults then rebuild. The sdkconfig.h file should then get rebuilt, taking into account UPLOAD_PORT.

AnthonyKNorman, 2018-11-05:
Thanks for that. By accident I found that building an application without debug achieved the same thing. Anyway, I am up and running now. Thanks.

closed 2018-12-02 by mkellner