Moddable-OpenSource/moddable #102: Chromebook and Christmas Lights

creationix, 2018-11-09:
A while back I bought several reels of WS2812 lights (aka neopixels).  These are little RGB leds that are individually addressable via a single digital wire protocol.  I also have a whole box of various ESP8266 boards.

Every Christmas, I flash the esp8266 boards with Espruino, hook them to the house wifi and let him have a ball programming animations over wifi using the IDE that's an app on his chromebook.

If you've not seen this workflow, it's pretty amazing.  He simply installs this [app](https://chrome.google.com/webstore/detail/espruino-web-ide/bleoifhkdalbjfbobjackfdifdneehpo) on his chromebook and connects to the esp8266 over serial or wifi.  He's got a collection of JS starter projects I wrote for him and tweaks them to write new animations.

[image]

I was wondering how much work it would be to enable this experience, but using moddable instead of espruino.  I want him to start learning modern JavaScript and espruino is a strict subset of ES5.1 with poor debugging capabilities.

Chrome apps can talk to serial devices as well as full access to local TCP or UDP networks.  They don't have a C compiler though, it would have to be ported to run in webassembly or something.

Can I build moddable in such a way that it enables sending JS source code over wifi or serial?  Also are there any existing drivers for WS2812 protocol?  If not, I've written them before in C, it's a fairly simple protocol, just very timing sensitive.

phoddie, 2018-11-09:
Fun. ;) Let's figure this out. My kids would happily play with this too...

We already have Neopixels working (WS2812) on ESP32. There's a [blog](http://blog.moddable.com/blog/neopixels/) post with the details and the obligatory blinking lights video. We can surely adapt an existing driver on ESP8266 to support the same API. If you want to write that from scratch, that's great too. The timing sensitives always made it look like a project that would take a while...

We can compile JavaScript directly on the device too. We don't generally do that for a bunch of good reasons, but it works. This [blog post](http://blog.moddable.com/blog/eval/) describes the details. Sending scripts over serial would be best, as that is more reliable than Wi-Fi. Also, serial is used by the xsbug. We do something like this already in the test262 tools to run the test suite on MCUs. It should be possible to adapt. I can look into that.

creationix, 2018-11-09:
Also if it's easier, I'm not opposed to finally updating to esp32.  Especially if this means I can serial over bluetooth and not need a wire.

phoddie, 2018-11-09:
It would be easier on ESP32 - lots more memory and Neopixels already works. Maybe let's start there. We have BLE working nicely on ESP32 haven't explored using it as a serial replacement yet. That's sounds like an entirely separate adventure...

phoddie, 2018-11-09:
@bfriedkin reminds me that we do have a [serial BLE example](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/ble/ble-friend/main.js) that works with the [Adafruit BLEFriend](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/ble/ble-friend/main.js). The speed may be too slow for reasonable use though. A Classic BLE serial connection might be more appropriate.