Moddable-OpenSource/moddable #103: ESP32 UART Support

AnthonyKNorman, 2018-11-19:
I can find no mention of a uart driver for ESP32. Am I right?

phoddie, 2018-11-19:
Check out this [contribution](https://github.com/Moddable-OpenSource/moddable/tree/public/contributed/serial) -- "Preliminary version of hardware Serial communication module for esp32."

mkellner, 2018-11-20:
Anthony, I'm working on a Serial module which I intend to use to talk to a GPS (u-blox 7) and Cell module (SIM7100). I've got something starting to show signs of life.

What kind of device are you intending to connect to?

AnthonyKNorman, 2018-11-20:
My first requirement is for a small MP3 player board that is driven from a Serial interface. So just fairly basic transmit and receive. Not particularly large amounts of data.

mkellner, 2018-11-20:
I've forked the tree and replaced the **contributed/serial** with a new one.
[mkellner branch - serial module](https://github.com/mkellner/moddable/tree/public/contributed/serial)

[And a little description](https://github.com/mkellner/moddable/blob/public/contributed/serial/SerialModule.md) of the API.

I wrote this to communicate with text-based serial devices so it works with strings. It may work for you. If it doesn't, please let me know what doesn't work properly and how you expect it to work.

In the next step for this project, I would like to add an asynchronous way of handling data reception.
Where you would add a callback to the serial object and you'd get called back when a string was received. Like:

```
    serial.onStringReceived(string) = {
       handleData(string);
    }
```

AnthonyKNorman, 2018-11-21:
Hi Michael
I'll give this a try. I like the callback idea.
I think, ideally, it could support writeBytes(arrayBuffer). It so happens that my application requires sending a byte-based protocol with start byte, end byte, checksum etc. Obviously I can encode this as a string, but it looks quite clumsy.
I'll let you know when I've been able to try your code.
Thanks for this.
Tony

mkellner, 2018-11-23:
I've pushed an update to the forked repository above.

Now serial.write() will take a string or an arrayBuffer and optional start and length parameters.

Also, there is a serial.poll command which starts a thread to read data. You can get called back when a terminator is read or when a specific number of bytes are read.

AnthonyKNorman, 2018-11-28:
I am using your fork with the arrayBuffer option to write to a little dfplayer mp3 player card. Works fine! I haven't tried the serial.poll yet, but it looks cool. I will be tweeting about my project in the next couple of days @ProjectPages

AnthonyKNorman, 2018-12-31:
Hi Michael
I have just returned to my dfPlayer MP3 project. As I said, playing a MP3 works fine. However, I now have the same problem with readBytes that I had with write. The player returns a block of binary data, which frequently contains zeroes. This means it is interpreted as a string terminator. Can you suggest a way around this, or could you maybe implement readBytesInto(arrayBuffer, bytes)? Retaining support for the terminator would be good, as this data always terminates with 0xEF, but that is not essential.
Thanks, Tony

mkellner, 2019-01-03:
Tony, have you tried `readBytesUntil(0xEF, maxLen);`
It should only look for the 0xEF terminator (as written above) and not the string terminator 0x00.
If this doesn't do what you need it to, please let me know.

AnthonyKNorman, 2019-01-03:
Hi Michael
Yes, I found I could read all the bytes, but, because a string is returned, any attempt to iterate through it stops as soon as it encounters the zero. I have solved my problem by collecting one byte at time, and, if an empty string is returned, assume the value is zero, but it's not pretty!
Tony

mkellner, 2019-01-08:
The fork https://github.com/mkellner/moddable  has been updated to match the latest Moddable drop. I've added the ability to use an array buffer in readBytes. There's a new example.

So this should now work:
let arrayBuffer = new ArrayBuffer(64);
serial.readBytesUntil(arrayBuffer, [0xEF,0]);

(the second parameter is a string of terminator characters).

mkellner, 2019-01-29:
Anthony, the OpenSource tree now contains the changes to the serial driver for esp32.

closed 2019-01-29 by mkellner