Moddable-OpenSource/moddable #104: Does not build esp32 samples

barbiani, 2018-11-25:
Windows 10 builds for the simulator, but not for the esp32 target.

"C:\Users\future\Documents\Moddable\build\tmp\esp32\moddable_zero\debug\xsProj" actually exists!

`C:\Users\future\Documents\Moddable\examples\helloworld>cd %MODDABLE%\examples\piu\balls

C:\Users\future\Documents\Moddable\examples\piu\balls>mcconfig -d -m -p esp32
Does C:\Users\future\Documents\Moddable\build\tmp\esp32\moddable_zero\debug\xsProj specify a file name
or directory name on the target
(F = file, D = directory)? d
C:\Users\future\Documents\Moddable\build\devices\esp32\xsProj\Makefile
C:\Users\future\Documents\Moddable\build\devices\esp32\xsProj\partitions.csv
C:\Users\future\Documents\Moddable\build\devices\esp32\xsProj\sdkconfig.defaults
C:\Users\future\Documents\Moddable\build\devices\esp32\xsProj\sdkconfig.defaults.prior
C:\Users\future\Documents\Moddable\build\devices\esp32\xsProj\sdkconfig.defaults.release
C:\Users\future\Documents\Moddable\build\devices\esp32\xsProj\main\component.mk
C:\Users\future\Documents\Moddable\build\devices\esp32\xsProj\main\main.c
7 File(s) copied
        1 file(s) copied.
FC: cannot open C:\USERS\FUTURE\DOCUMENTS\MODDABLE\BUILD\DEVICES\ESP32\XSPROJ\SDKCONFIG.DEFAULTS.PRIOR - No such file or folder

        1 file(s) copied.
The system cannot find the path specified.
NMAKE : fatal error U1077: 'if' : código de retorno '0x1'
Stop.`

bfriedkin, 2018-11-27:
Hello -

Thank you for the follow-up and additional notes. It is helpful to know that you have the ESP-IDF installed in a different place.

The Moddable ESP32 build requires a number of environment variables to be set. The ESP32 setup for Windows documented in our [Getting Started](https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/Moddable%20SDK%20-%20Getting%20Started.md#esp32-windows) instructions covers what is required. My hunch is that one or more of the environment variables might not be properly set. In particular, the IDF_PATH environment variable needs to be set in the MSYS2 environment using forward slashes. See step 7 in the setup instructions. If you additionally set %IDF_PATH% in the Windows environment, that %IDF_PATH% should specify a Windows-style directory path, e.g. c:\Sloeber\arduinoPlugin\packages\esp32\esp-idf.

The Windows all-in-one toolchain (see step 3) you downloaded contains the GCC toolchain used to cross-compile the code. This GCC toolchain is used to build both the Moddable runtime and the ESP32 firmware. The mcconfig tool calls NMAKE on Windows to execute the nmake.esp32.mk make file. The make file builds the the Moddable runtime and launches a MSYS2 window to build the ESP32 firmware. Everything is built using command line tools. It is important to follow the prompts in both the Developer Command Prompt and MSYS2 windows.

My suggestion is to first try and get the ESP32 build working by following the Getting Started instructions. Steps 1 through 10 are one-time setup. Once that works, you can revisit the build failure issue after seeing how a successful build works.

As for the "fatal error", based on the console output provided, I am wondering if the problem is here at line 282 in nmake.esp32.mk:

if exist $(TMP_DIR)\_s.tmp ($(MSYS32_BASE)\msys2_shell.cmd -mingw32 -c "echo Running GENCONFIG...; BATCH_BUILD=1 DEBUG=$(DEBUG) IDF_BUILD_DIR=$(IDF_BUILD_DIR_MINGW) SDKCONFIG_DEFAULTS=$(SDKCONFIG_FILE_MINGW) make defconfig")

If the MSYS32_BASE environment variable is not correct, the make file won't be able to find the msys2_shell.cmd executable. At the top of the nmake.esp32.mk file, you can follow how the various environment variables are used by the make file.

Regards,
Brian

barbiani, 2018-11-28:
I gave up on windows and went back to my ubuntu vm.

Had to move the esp32 and idf compiler to $HOME/esp32/... idf examples find the toolchain through the PATH variable, but moddable does not.

Got the idf 3.1 hello-world example building.

Built xsbug and the emulator.

cd ${MODDABLE}/examples/helloworld
mcconfig -d -m -p lin <-- goes fine and opens the emulator

box:~/Downloads/moddable/examples/helloworld$ mcconfig -d -m -p esp32
# cc xsHost.c (strings in flash)
In file included from /home/future/Downloads/moddable/xs/sources/xsCommon.h:41:0,
                 from /home/future/Downloads/moddable/xs/sources/xsAll.h:41,
                 from /home/future/Downloads/moddable/xs/platforms/esp/xsHost.c:38:
/home/future/Downloads/moddable/xs/platforms/esp/xsPlatform.h:100:24: fatal error: lwip/netdb.h: No such file or directory
compilation terminated.
make: *** [/home/future/Downloads/moddable/build/tmp/esp32/debug/helloworld/makefile:401: /home/future/Downloads/moddable/build/tmp/esp32/debug/lib/xsHost.c.o] Error 1

Does not seem to find the lwip idf component by itself.

netdb.h is in ~/esp32/esp-idf/components/lwip/lwip/src/include/lwip/ (IDF 3.1).

bfriedkin, 2018-11-28:
Hello -

Please make sure your local Moddable repository is up-to-date and double check that you are building against v3.1 of the ESP-IDF. We recently updated our ESP32 build to be compatible with the v3.1 ESP-IDF. The 3.1 ESP-IDF includes multiple lwip directories.

Our ESP32 build make.esp32.mk make file looks for lwip include files in these ESP-IDF base directories (lines 65-66):

 	$(IDF_PATH)/components/lwip/include/lwip \
 	$(IDF_PATH)/components/lwip/include/lwip/port \

Regards,
Brian

barbiani, 2018-11-28:
Brian,

Cloned the 3.1 branch specifically.

First time build  threw error at uecc and bootloader. Ran mcconfig again. It flashed my esp32 and opened  xsbug. Hello world example is now running!

Thank you very much.

closed 2018-11-28 by bfriedkin