Moddable-OpenSource/moddable #105: Custom module resolution using public headers only?

creationix, 2018-11-28:
Related to #91, I was wondering if there was a way to register custom module resolution  / loading functions without using internal APIs?  As I explained in #79, using the internal private headers is problematic for me.  But as far as I can tell, customizing the module loader isn't currently possible via the public headers.

lll000111, 2019-03-11:
@creationix If this still is an issue, since I faced the same problem (issues #135, #138), my solution was to rewrite the paths in my JS files' `import` statements and even to move files around during the build process that I needed for the JS files anyway to strip away Flow types.

Now my other platforms use a directory based approach and when building for Moddable files are moved &mdash; because especially `./` as path is tricky, since it means something different depending on where the importing file lives &mdash; and paths replaced. On Moddable I use plain-name imports (no path) mostly, and a single word path preface for libs. Within the lib(s), which uses relative path `./` and `../` a lot, I had to shift files around, making their subdirectory instead path of their name. For example, what was `./util/function-helpers.js` now is `./util-function-helpers.js'. Of course, tools like type checkers won't work on that code any more after the built because the imports no longer point to directories, but they still work for the true source code so who cares.

I find that this solution isn't actually all that complicated, and you have to think about it and implement it only once since it's unlikely that a project's directory structure changes a lot.

creationix, 2019-04-26:
I've not been using Moddable for a while, we switched to V8 at my day job.  But FYI node 12 just launched with module support and it also matches browser semantics with relative paths and extensions.  Sounds like this would be a good default behavior for modules out of the box.

phoddie, 2019-05-13:
@creationix - we are taking a look at this as part of a larger investigation. That said, the runtimes are wildly different, so some differences are to be expected.

closed 2019-05-13 by phoddie