Moddable-OpenSource/moddable #112: JSON.stringify correctness issue

mathiasbynens, 2019-01-04:
```
$ eshost -x 'var allString = true; JSON.stringify([1,2,3,4,5], function(k,v){allString = allString && (typeof k == "string"); return v}); print(allString)'
#### Chakra
true

#### JavaScriptCore
true

#### SpiderMonkey
true

#### V8
true

#### V8 --harmony
true

#### XS
false
```

mathiasbynens, 2019-01-04:
Another one:

```
$ eshost -x 'var replaceTracker; function replaceFunc(key, value) { replaceTracker += key + "("+(typeof key)+")" + JSON.stringify(value) + ";"; return value; } replaceTracker = ""; JSON.stringify([1,2,3,,,,4,5,6], replaceFunc); print(replaceTracker);'
#### Chakra
(string)[1,2,3,null,null,null,4,5,6];0(string)1;1(string)2;2(string)3;3(string)undefined;4(string)undefined;5(string)undefined;6(string)4;7(string)5;8(string)6;

#### JavaScriptCore
(string)[1,2,3,null,null,null,4,5,6];0(string)1;1(string)2;2(string)3;3(string)undefined;4(string)undefined;5(string)undefined;6(string)4;7(string)5;8(string)6;

#### SpiderMonkey
(string)[1,2,3,null,null,null,4,5,6];0(string)1;1(string)2;2(string)3;3(string)undefined;4(string)undefined;5(string)undefined;6(string)4;7(string)5;8(string)6;

#### V8
(string)[1,2,3,null,null,null,4,5,6];0(string)1;1(string)2;2(string)3;3(string)undefined;4(string)undefined;5(string)undefined;6(string)4;7(string)5;8(string)6;

#### V8 --harmony
(string)[1,2,3,null,null,null,4,5,6];0(string)1;1(string)2;2(string)3;3(string)undefined;4(string)undefined;5(string)undefined;6(string)4;7(string)5;8(string)6;

#### XS
(string)[1,2,3,null,null,null,4,5,6];0(number)1;1(number)2;2(number)3;3(number)undefined;4(number)undefined;5(number)undefined;6(number)4;7(number)5;8(number)6;
```

patrick-soquet, 2019-01-05:
Thank you much for the report. I committed a fix which should be published next week. Meanwhile xst 8.6.2 is available so you can test with eshost.

phoddie, 2019-01-29:
The fix is now committed.

closed 2019-01-29 by phoddie