Moddable-OpenSource/moddable #114: Error while running hello world on moddable zero

getachew, 2019-01-11:
Hi Moddable community,

I went through the Getting Started documentation and install all the required dependencies as per the guide. But I am running into this error when running the hello world example app. I feel like I am missing a step but curious if anyone has had this error and resolved it

```
$ cd ${MODDABLE}/examples/helloworld
$ mcconfig -d -m -p esp
# ld main.bin
/Users/it/Projects/moddable/build/tmp/esp/debug/lib/xsPlatform.c.o:(.text.fxConnect+0x38): undefined reference to `espConnectToWiFi'
/Users/it/Projects/moddable/build/tmp/esp/debug/lib/xsPlatform.c.o: In function `fxConnect':
/Users/it/Projects/moddable/xs/platforms/esp/xsPlatform.c:258: undefined reference to `espConnectToWiFi'
collect2: error: ld returned 1 exit status
make: *** [/Users/it/Projects/moddable/build/bin/esp/debug/helloworld/main.bin] Error 1
$ 
```

phoddie, 2019-01-11:
Whoops. It looks like you did everything correctly. That call to `espConnectToWiFi`  shouldn't be there. In $MODDABLE/xs/platforms/esp/xsPlatform.c  just comment out the call to `espConnectToWiFi`. We'll push a fix for that. Thanks for reporting the problem.

phoddie, 2019-01-11:
Fixed now. Thanks, @mkellner .

closed 2019-01-11 by phoddie