Moddable-OpenSource/moddable #119: Cannot build example/js/helloregexp/main.js for esp

horihiro, 2019-02-03:
I got following error when building %MODDABLE%/example/js/helloregexp/main.js for esp (not esp32) on Windows.

```
moddable\examples\js\helloregexp>mcconfig -d -m -p esp
# xsc time.xsb
# xsc timer.xsb
# xsc Resource.xsb
# xsc instrumentation.xsb
# xsc main.xsb
# xsc mc\config.xsb
# xsid modTime.c.xsi
# xsid modTimer.c.xsi
# xsid timer.c.xsi
# xsid Resource.c.xsi
# xsid modInstrumentation.c.xsi
# xsid modTimer.h.xsi
# xsid modInstrumentation.h.xsi
# xsl modules
# mcrez resources
Total resource size: 0 bytes
# cc modTime.c.o (strings in flash)
# cc modTimer.c.o (strings in flash)
# cc timer.c.o
# cc Resource.c.o (strings in flash)
# cc modInstrumentation.c.o (strings in flash)
# cc xsHost.o
# cc xsPlatform.o
# cc main.o
# archive C:\Users\horihiro\Projects\moddable\build\bin\esp\debug\helloregexp\libxsar.a
# ld main.bin
C:\Users\horihiro\Projects\moddable\build\tmp\esp\debug\lib/libxslib.a(xsRegExp.o):(.text.fxInitializeRegExp+0x4): undefined reference to `fxCompileRegExp'
C:\Users\horihiro\Projects\moddable\build\tmp\esp\debug\lib/libxslib.a(xsRegExp.o): In function `fxInitializeRegExp':
C:\Users\horihiro\Projects\moddable\xs\sources/xsRegExp.c:75: undefined reference to `fxCompileRegExp'
C:\Users\horihiro\Projects\moddable\build\tmp\esp\debug\lib/libxslib.a(xsRegExp.o): In function `fxExecuteRegExp':
C:\Users\horihiro\Projects\moddable\xs\sources/xsRegExp.c:75: undefined reference to `fxMatchRegExp'
C:\Users\horihiro\Projects\moddable\build\tmp\esp\debug\lib/libxslib.a(xsRegExp.o): In function `fx_RegExp_prototype_exec':
C:\Users\horihiro\Projects\moddable\xs\sources/xsRegExp.c:75: undefined reference to `fxMatchRegExp'
collect2: error: ld returned 1 exit status
NMAKE : fatal error U1077: 'C:\Users\horihiro\esp\xtensa-lx106-elf\bin\xtensa-lx106-elf-g++.EXE' : リターン コード '0x1'
Stop.
```

Could you please help me to build esp app which uses RegExp on Windows?

bfriedkin, 2019-02-04:
@horihiro  -

Thank you for reporting this issue. As you suggested, the Windows build is missing a required file.

To resolve the issue, please edit your `%MODDABLE%\tools\mcconfig\nmake.esp.mk` make file and add the required `xsre.o` file here:

```
	$(LIB_DIR)\xsmc.o \
	$(LIB_DIR)\xsre.o
```

We will include this fix in our next Moddable SDK Open Source update.

Regards,
Brian

closed 2019-02-04 by bfriedkin

horihiro, 2019-02-04:
Thanks @bfriedkin !

I think that `xsBigInt.o` is needed to add into `nmake.esp.mk` in order to work another example ( `%MODDABLE%/examples/network/http/httpget/main.js` ) for esp on Windows.

```
	$(LIB_DIR)\xsAtomics.o \
	$(LIB_DIR)\xsBigInt.o \
	$(LIB_DIR)\xsBoolean.o \
```

Could you please confirm this too?

Regards,

bfriedkin, 2019-02-04:
@horihiro -

We apologize for the inconvenience. Attached please find a copy of nmake.esp.mk that should include all the missing files.

Please let us know if you bump into any other issues.

Regards,
Brian

[nmake.esp.mk.zip](https://github.com/Moddable-OpenSource/moddable/files/2830009/nmake.esp.mk.zip)

horihiro, 2019-02-05:
Thanks @bfriedkin 

I will try the file you attached.
If I find other issues, I will file new issue.

And you don't need to apologize, we appreciate for this awesome SDK.

Best Regards,