Moddable-OpenSource/moddable #120: Glitch on piu and M5Stack

meganetaaan, 2019-02-05:
I compiled a simple piu code for M5Stack with latest commit (dc94381c29cfe76dc7e565885a2ff1eff8c501f7).
But I see a glitch on the display.

---

main.js
```JavaScript
import {} from "piu/MC";

export default new Application(null, {
  skin: new Skin({ fill: "blue" }),
  top: 0,
  bottom: 0,
  left: 0,
  right: 0
});
```

manifest.json

```json
{
  "include": [
    "$(MODDABLE)/examples/manifest_base.json",
    "$(MODDABLE)/examples/manifest_piu.json"
  ],
  "modules": {
    "*": "./main"
  }
}
```

```bash
$ mcconfig -d -m -p esp32/m5stack
```

Expected -> Whole screen is blue as I set.
Actual -> A vertical black belt shows up on the display like below.
[image]

No problem on the same code with the simulator. So is there an issue of ili9341 driver?
Or is my program wrong?

```bash
$ mcconfig -d -m -p lin
```

[image]

phoddie, 2019-02-05:
@meganetaaan - since this works on the simulator and almost works on the target hardware, your intuition is likely correct: this is likely related to the video controller in some way. We have noticed some inconsistencies with the screen on the M5Stack. Perhaps there are some variations in the components used, for example.

You could try slowing down the SPI transfer. We are driving the ILI9341 faster than the data sheet recommends. In `$MODDABLE/build/devices/esp32/target/m5stack/manifest.json` you can do that here:

		"ili9341": {
			"hz": 40000000,

As a test, reduce the 40000000 to 10000000.

If that does not help. it may be necessary to change the register initialization. That is a little more difficult. If you have an example that works using a different software stack, we can extract the register settings from that to use. Please let me know.

meganetaaan, 2019-02-06:
Thank you for your quick response!

I've tried changing the spi transfer rate to 10000000 

```json
	"ili9341": {
		"hz": 10000000,
```

Then the glitch shrunk some but still shown.

> If you have an example that works using a different software stack, we can extract the register settings from that to use.

I usually use the M5Stack library for Arduino. It depends on a modified version of [TFT_eSPI](https://github.com/Bodmer/TFT_eSPI). Here is [Initialization sequence](https://github.com/m5stack/M5Stack/blob/0ae59089b93585c3f4b21ef7c263039571f8f421/src/utility/In_eSPI.cpp#L222).

I've noticed this library is [set up with 40000000 hz frequency](https://github.com/m5stack/M5Stack/blob/0559d05ab5cfcbea5f314df55789fdf4c856e0c0/src/utility/In_eSPI_Setup.h#L202) as Moddable do. So I guess it's not a main cause.

In addition, I can successfully run some Moddable examples ([text](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/text), [bars](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/bars) and [neon-light](https://github.com/Moddable-OpenSource/moddable/tree/public/examples/piu/neon-light)) as-is.

Hope this helps.

phoddie, 2019-02-06:
@meganetaaan - thanks for trying the SPI change.

I applied the full register set from the initialization sequence you reference to the m5stack/manifest.json. On the hardware I have here (M5 Stack Fire), the result appears the same - the piu/balls example runs perfectly. Maybe we have different display controllers?

Here is the relevant change to the [manifest](https://github.com/Moddable-OpenSource/moddable/blob/public/build/devices/esp32/targets/m5stack_fire/manifest.json#L16).

`````
		"ili9341": {
			"hz": 40000000,
			"cs_pin": 14,
			"dc_pin": 27,
			"rst_pin": 33,
			"spi_port": "VSPI_HOST",
			"backlight_pin": 32,
			"backlight_on": 1,
			"registers": [
				"0x01, 1, 0xA8,",
				"0xFF, 5,",
				"0xEF, 3, 0x03, 0x80, 0x02,",
				"0xCF, 3, 0x00, 0XC1, 0X30,",
				"0xED, 4, 0x64, 0x03, 0X12, 0X81,",
				"0xE8, 3, 0x85, 0x00, 0x78,",
				"0xCB, 5, 0x39, 0x2C, 0x00, 0x34, 0x02,",
				"0xF7, 1, 0x20,",
				"0xEA, 2, 0x00, 0x00,",
				"0xC0, 1, 0x23,",
				"0xC1, 1, 0x10,",
				"0xC5, 2, 0x3e, 0x28,",
				"0xC7, 1, 0x86,",
				"0x36, 1, 0xA8,",
				"0x3A, 1, 0x55,",
				"0xB1, 2, 0x00, 0x13,",
				"0xB6, 3, 0x08, 0x82, 0x27,",
				"0xF2, 1, 0x00,",
				"0x26, 1, 0x01,",
				"0xE0, 15, 0x0F, 0x31, 0x2B, 0x0C, 0x0E, 0x08, 0x4E, 0xF1, 0x37, 0x07, 0x10, 0x03, 0x0E, 0x09, 0x00,",
				"0xE1, 15, 0x00, 0x0E, 0x14, 0x03, 0x11, 0x07, 0x31, 0xC1, 0x48, 0x08, 0x0F, 0x0C, 0x31, 0x36, 0x0F,",
				"0x11, 0,",
				"0xFF, 120,",
				"0x29, 0,",
				"0xFF, 0,",
			],
		},
``````

Would you give it a try?

meganetaaan, 2019-02-07:
I tried changing my manifest.json as above. It works like a charm! Thank you so much!
Now the glitch has completely disappeared.

I don't have M5Stack Fire but M5Stack basic. But above changes are valid the same.

I tried extra experiment to reduce those register declaration and see if there's any unnecessary.
Then I found that only `"0xFF, 120"` (120ms delay) is enough to eliminate glitch. The wait time can be even short as 30ms.

My manifest.json is below after all.

```
    "ili9341": {
      "hz": 40000000,
      "cs_pin": 14,
      "dc_pin": 27,
      "rst_pin": 33,
      "spi_port": "VSPI_HOST",
      "backlight_pin": 32,
      "backlight_on": 1,
      "registers_append": ["0x36, 1, 0xA8,", "0xFF, 30,"]
    },
```

phoddie, 2019-02-08:
@ meganetaaan - Thanks for confirming the change and for finding a more concise change. Since that works for both M5 Stack Basic and Fire, I'll commit that. It will included in our next update, probably next week.

closed 2019-02-08 by phoddie