Moddable-OpenSource/moddable #122: Unspecific mcconfig error

lll000111, 2019-02-09:
Running examples from the examples folder worked. Then I created a directory somewhere else and copied the manifest files from examples/ plus one main.js and manifest.json from a simple example into it.

Now I get

```
mha@mint-virtual-machine ~/Projects/one.moddable $ mcconfig -d -m -p lin
# error: File unsupported!
### Error: incompatible platform!
```

Which file is unsupported? Which platform is incompatible? As far as I can see it's the exact same manifest files and example file, only that I copied them to another place outside the moddable/ directory. Too bad the error messages are missing specific information about what's wrong.

phoddie, 2019-02-09:
It refers to the `File` module, hence the capital "F" in file. If you search for "[File supported](https://github.com/Moddable-OpenSource/moddable/blob/dc94381c29cfe76dc7e565885a2ff1eff8c501f7/modules/files/file/manifest.json#L23)" in the repository you will find it easily.

closed 2019-02-09 by phoddie

lll000111, 2019-02-09:
May I suggest putting that into ""?

I edited a lot of my own error messages along the same lines after realizing this exact problem: Writing word and writing "word" are very different, the quotes provide crucial hints for reading the sentence.

phoddie, 2019-02-11:
Thank you for the suggestion.. We've added the word `module` to theese messages, e.g. `File module unsupported`, for the next update.

lll000111, 2019-02-15:
A quick question for clarification:

Apparently file system support is _only_ available on ESP 8266 but not on ESP 32 (or anything else)?

phoddie, 2019-02-15:
Yes, the build output that the `File` module is unsupported for the ESP32 target is accurate.

The implementation for ESP32 would be quite similar if, like the ESP8266 implementation, it also uses the SPIFFS API and quite different if it uses the file APIs in the ESP IDF instead.

lll000111, 2019-02-15:
Okay... so do you guys have any plans for it? We may or may not do it if you won't, not sure yet. I'm certainly not expecting anything for nothing, just asking.

phoddie, 2019-02-15:
> ...do you guys have any plans for it?

There are no immediate plans to support File on ESP32. I expect it will happen, one way or another, this calendar year.

lll000111, 2019-02-15:
Ok, good to know, thanks!