Moddable-OpenSource/moddable #124: documentation/network/network.md link to examples leads to 404

lll000111, 2019-02-11:
Page: https://github.com/Moddable-OpenSource/moddable/blob/public/documentation/network/network.md#class-websocket-client

Link: 

> **Relevant Examples:** [websocketclient](https://github.com/Moddable-OpenSource/moddable/blob/public/examples/network/websocket/websocketclient)

   404 Page Not Found

There is not even a folder `examples/network/websocket/`

lll000111, 2019-02-11:
By the way, looking through the code for websocket client/server and the examples in order to write Flow type definitions for it, are you sure this works?

For example, the documentation page says `Server` has a method `write`. I don't see one in the code.

phoddie, 2019-02-11:
> By the way, looking through the code for websocket client/server and the examples in order to write Flow type definitions for it, are you sure this works?

Yes.

> For example, the documentation page says `Server` has a method `write`. I don't see one in the code. Write methods [seem to be on the `Client` created in the `Server` constructor for each new request](https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/websocket/websocket.js#L273).

The documentation you reference describes how to use the module. It does not document the implementation.

lll000111, 2019-02-11:
> The documentation you reference describes how to use the module. It does not document the implementation.

Well what great news /s

I don't see where there's a write method on `Server` instances: https://github.com/Moddable-OpenSource/moddable/blob/public/modules/network/websocket/websocket.js#L270

phoddie, 2019-02-12:
> I don't see where there's a `write` method on `Server` instance

The `Server` is a listener waiting for incoming connections. It is not a TCP connection itself, so it is meaningless for it have a `write` function as there is no destination for the data. When the server receives a connection request, it accepts the request and creates a client instance. The client instance has a `write` function which is used to transmit data to the host at the other end of the connection.

lll000111, 2019-02-12:
I know that, but please look at the documentation. It documents a `write` function that does not exist, but on the other hand it is silent about the listener stuff which is how the server actually works, as you write.

I'm not stupid and not a newbie. If I have a lot of "beginner questions" it's because not everything is clear from the docs.

phoddie, 2019-02-12:
I understood that you were attempting to use the WebSocket code in a project. I have tried to help you with that. I did not understand that you were asking a question about the documentation.

lll000111, 2019-02-12:
Yeah sorry, context switch-

lll000111, 2019-02-12:
Actually, the whole original submission was about documentation

And I wrote "For example, **the documentation** page says `Server` has a method `write`."

:-)

phoddie, 2019-02-15:
> the whole original submission was about documentation

The issue identified in the original report will be corrected with the next update to the repository. Thank you.

lll000111, 2019-03-08:
It was solved.

closed 2019-03-08 by lll000111